platformio/platform-ststm32 #1: mbed RTOS not building on board disco_f051r8

carvalhais, 2016-08-09:
Tinkering with the examples for mbed platform found in [here](https://github.com/platformio/platformio-examples/tree/develop/mbed), I get the following error when trying to compiled the mbed-rtos example:

```
[Tue Aug  9 10:43:04 2016] Processing disco_f051r8 (platform: ststm32, board: disco_f051r8, framework: mbed)
--------------------------------------------------------------------------------
arm-none-eabi-g++ -o .pioenvs/disco_f051r8/firmware.elf -Wl,--gc-sections -Wl,--wrap,main -mcpu=cortex-m0 -mthumb --specs=nano.specs -Wl,-T"/home/carvalhais/.platformio/packages/framework-mbed/variant/DISCO_F051R8/mbed/TARGET_DISCO_F051R8/TOOLCHAIN_GCC_ARM/STM32F0xx.ld" .pioenvs/disco_f051r8/src/main.o -L/home/carvalhais/.platformio/packages/ldscripts -L/home/carvalhais/.platformio/packages/framework-mbed/variant/DISCO_F051R8/mbed/TARGET_DISCO_F051R8/TOOLCHAIN_GCC_ARM -L/home/carvalhais/.platformio/packages/framework-mbed/libs/rtos/TARGET_DISCO_F051R8/TOOLCHAIN_GCC_ARM -L/home/carvalhais/.platformio/packages/frameworkmbed/libs/rtos/TARGET_DISCO_F051R8/TOOLCHAIN_GCC_ARM -L.pioenvs/disco_f051r8 -Wl,--start-group -Wl,-whole-archive .pioenvs/disco_f051r8/libFrameworkMbed.a -Wl,-no-whole-archive -lsupc++ -lmbed -lc -lgcc -lc -lgcc -lm -lstdc++ -lnosys -lrtos -lrtx -Wl,--end-group
/home/carvalhais/.platformio/packages/framework-mbed/libs/rtos/TARGET_DISCO_F051R8/TOOLCHAIN_GCC_ARM/librtx.a(HAL_CM0.o): In function `SysTick_Handler':
HAL_CM0.S:(.text+0x12c): multiple definition of `SysTick_Handler'
.pioenvs/disco_f051r8/libFrameworkMbed.a(system_stm32f0xx.o):system_stm32f0xx.c:(.text.SysTick_Handler+0x0): first defined here
collect2: error: ld returned 1 exit status
scons: *** [.pioenvs/disco_f051r8/firmware.elf] Error 1
========================= [ ERROR ] Took 0.48 seconds =========================
```

However, if I try the mbed-blink example, the code compiles without any issue, which suggests the error might be in the "rtos.h" file.

It seems this is a linker issue confirmed by other users, as it can be seen in this community thread [here](https://community.platformio.org/t/unable-to-compile-blink-led-example-with-mbed-rtos/595).

I'm opening this ticket on behalf of valeros, as he asked in the link above.

ivankravets, 2016-08-25:
Could you try new build system for ARM mbed? See instruction here:
- https://community.platformio.org/t/platformio-3-0-improved-integration-with-arm-mbed-ecosystem/639

ivankravets, 2016-10-11:
Please reopen if you still have issue with new PlatformIO 3.0

closed 2016-10-11 by ivankravets