platformio/platform-ststm32 #3: Add support for STM32  Nucleo board - L432KC (and ST Nucleo L011K4)

mcprakash, 2016-09-08:
It would be great to have platformio support for the  [Nucleo board (L432KC)](https://developer.mbed.org/platforms/ST-Nucleo-L432KC/). 
At least a schematic of how to add new boards of the same family would be really helpful.

The other boards of the same Nucleo family that are supported are listed below:
1. ST Nucleo F303K8
2. ST Nucleo F031K6 
3. ST Nucleo F042K6
4. ST Nucleo L031K6 

Also another missing one from the same family- ST Nucleo L011K4

Thanks 

Chandra

closed 2016-09-13 by ivankravets

Davey3D, 2017-06-17:
stm32-l432 supported for Arduino yet?

mcprakash, 2017-06-17:
@Davey3D 
  STM32 works for mbed-os. Don't think it is supported for Arduino environment, but it should be fairly easy to add. Please follow the reference manual for adding a new board to an environment. 

Thanks

Davey3D, 2017-06-17:
STM32  is supported in the Arduino IDE : 

http://www.stm32duino.com/
https://community.st.com/community/stm32-community/blog/2017/06/02/stm32-cores-enabled-in-arduino-ide
https://github.com/GrumpyOldPizza/arduino-STM32L4

mcprakash, 2017-06-17:
@Davey3D 
 I know that STM32 has an arduino interface. I have never personally used it using Arduino code interface but only with mbed-os interface. Please try "board = stm32" and "os = arduino" and let us know if it works.

Davey3D, 2017-06-17:
I'm starting to realize that I'm totally lost! lol I think my confusion is stemming from the discussion linked below and some fuzziness in terms. I asked the question there as well. 

My fuzzy logic went something along the lines of: platformio supports stm32-l432 boards and "Arduino" "Arduino" supports stm32-l432 boards. Therefore somehow I should be able to  program my stm32-l432 using Arduino sketches. Confusion being that "Arduino" is used to refer to boards, platforms, frameworks, and IDE's if I understand correctly. 

 github. com/ platformio /platform-ststm32/ issues/28 (spaces added to link)

 Let me ask a different way. I'd like to import my Arduino sketches into platformio and compile it to run on a stm32-l432 .   What part and I'm missing and who/what/how would I get it?  It looked promising as platformio lets me select the stm32-l432 board and select an option to maintain the code arduino. It generated:
     [env:nucleo_l432kc]
     platform=ststm32
     board=nucleo_l432kc
     framework=mbed

Compiling Arduino Blink gives error : .pioenvs/nucleo_l432kc/FrameworkMbed-platform-4e964/mbed_retarget.o: In function `__wrap_main':
mbed_retarget.cpp:(.text.__wrap_main+0xa): undefined reference to `main'

Changing to "framework=Arduino" Error: This board doesn't support arduino framework!
Changing to "platform= arduino" Error: Detected unknown package 'Arduino'
Changing to "board=stm32" is unknown board.

Maybe it's not possible and I'm stuck with Arduino IDE <gag> :-(

Thanks!

mcprakash, 2017-06-17:
Dude, 

Refer here -> http://platformio.org/frameworks/arduino
As you see, framework = Ardiuno should work. If it doesn't work, my best guess would be to use tool-stm32duino from here -> http://platformio.org/platforms/ststm32

If not, @ivankravets is the best person to answer. 

Thanks

Davey3D, 2017-06-18:
The package at that link only supports  STM32-F series boards. I have a STM32-L which is why I installed this package as it claims STM32L. 
https://github.com/GrumpyOldPizza/arduino-STM32L4

I went ahead and installed the files at your link just in case. Same error:

platformio.ini
  =====================
[env:nucleo_l432kc]
platform=ststm32
board=nucleo_l432kc
framework=Arduino
 
[platformio]
src_dir=Blink2
 
---------------------------------------
platformio run
l432kc; framework: Arduino)
 
Verbose mode can be enabled via `-v, --verbose` option
Error: This board doesn't support Arduino framework!

valeros, 2017-06-19:
Hi @Davey3D !
Unfortunately, there are few boards with Arduino support mainly because there is a plenty of different Arduino implementations for STM32 with different families. Our contributor  @ubis  decided to add support for this [one](https://github.com/rogerclarkmelbourne/Arduino_STM32). All STM32 boards with Arduino support you filter here http://platformio.org/boards

Davey3D, 2017-06-22:
You mean I blew $6 on an unsupported board!? LOL It'll make a nice necklace :-) I've googled a bit but not finding much on adding/changing frameworks. I'm guessing that means it's not an easy undertaking. It's either that or get another $3 out of the piggybank for a supported board :-)

Thanks to everyone for the personal and community support :-)

ubis, 2017-06-22:
@Davey3D There is an experimental/initial support for STM32L4, for the following boards:
> Butterfly-L433CC, Dragonfly-L476RE, Ladybug-L432KC, ST Nucleo L432KC, ST Nucleo L476RG

in my [repo](https://github.com/ubis/platform-ststm32). As you commented in another issue, you installed this repo, and still it didn't worked. Well, probably you installed the develoment branch. You actually need to install **stm32l4_support** repo.

I'm free now, so if we could move our conversation to my repo(create an issue) and solve it there, you might see official L4 support for these boards soon ;)

paulw517, 2018-07-11:
ubis, I just purchased three Nucleo L432KC boards to replace Arduino Mega boards for compactness.

I would like to enlist your assistance in the process to add a Nucleo L432KC board to the Arduino IDE.
Thank you
Paul Wilkie

paulw517, 2018-07-11:
Okay, I am following the instructions for installing the driver's from Zadig's download.  Where is the "Boot" button on the board?  The reset button is on the end of the board, button I do not see a "Boot" button??

I this referring to the Burn Bootloader button on the IDE?

paulw517, 2018-07-11:
I am at the stage of the GrumpyOldPizza instructions under OS Specifics - Windows - OS Specific Setup

I have downloaded the Zadig.exe and I am performing the STM32 Bootloader driver setup for Tera Corp boards.

I assume the boot button is the Burn Bootloader button on the IDE?  I pushed the boards reset button while highlighting the IDE’ Burn Bootloader button and the IDE returns the following error;

“Error while burning Bootloader”

Help?????

mcprakash, 2018-07-11:
Hi @paulw517 
 Can you online [mbed compiler](https://os.mbed.com/compiler/)  and see if "Blink LED" program works ? This will give an idea on we are with your setup. 

Thanks
Chandra

ubis, 2018-07-12:
@paulw517 This board seems to be supported by official STM32 Arduino core.
You can read my comment [here](https://github.com/platformio/platform-ststm32/issues/76) to try it.

PR #7 (2016-09-13): Add board ST-Nucleo-L432KC for mbed framework
@ivankravets 
 
I really want to add this board and start using it. Please let me know if I need to add some more configurations. 

This PR closes #3

Best

Chandra