platformio/platform-ststm32 #5: Two board configs with same mcu

ubis, 2016-09-11:
Alright, I noticed that there is two board configs with same mcu. However, they have some different options, and one is using mbed framework and another one is arduino.

As I can see, there are a few options:

- keep configs as it is now - there might be possible board config duplicates in the near future...
- keep one config for one framework, if another framework is used, make these config changes inside builder script - I think it should be a better way to do this...
- use one config only, change one framework directory structure so it would be similar to another one - due this, it should be maintained when there is any changes on original framework's repo.
- ???

This is only for one board yet, but if I recall correctly, stm32duino has some of stm nucleo boards, and there is already config's for these nucleo boards. So option one won't work here, and option two - builder script will be messy.

Bellow are the configs of boards.

[bluepill_f103c8.json](https://github.com/platformio/platform-ststm32/blob/develop/boards/bluepill_f103c8.json)
[genericSTM32F103C8.json](https://github.com/platformio/platform-ststm32/blob/develop/boards/genericSTM32F103C8.json)

ivankravets, 2016-09-12:
@zgoda , you pulled request with BluePill F103C8 board. I see that you filled some Arduino-based fields that are not used by mbed. Is this true? Could we merge to boards into one?

ARM mbed has own build flags, board settings and etc. We just link our Board ID with mbed TARGET in special [map](https://github.com/platformio/builder-framework-mbed/blob/develop/mbed.py#L48).

zgoda, 2016-09-13:
I used NUCLEO-F103RB as reference and my definition is for mbed, did not add any field nor removed any. Name is the same as used in mbed (ARM guys asked specifically to not use word "generic"). The only download protocol supported by mbed for this boards is stlink and STMicro suggests to use exactly the same ldscript as for NUCLEO-F103RB. That's what I see apart of framework-related stuff.

ivankravets, 2016-09-13:
Thanks, guys! I propose to leave it as it is. We can switch settings "on-the-fly". See in action https://github.com/platformio/platform-ststm32/blob/1abbab0122d0fbbabc608249b20fe4db4931a487/builder/frameworks/arduino.py#L33

commit 1abbab01 (2016-09-13): Add support for Arduino framework for BluePill F103C8 board // Resolve #5
closed 2016-09-13 by ivankravets