platformio/platform-ststm32 #6: mbed version 125 is released

evark, 2016-09-12:
Please update the mbed sources to last version

ivankravets, 2016-09-12:
Could you try to test new version manually? Does it work? For details, please check `~./platformio/packages/framework-mbed/platformio/README.txt`

ivankravets, 2016-09-12:
@zgoda Could you test new 125? Thanks.

evark, 2016-09-12:
Which is the best way to make the test? Which is your suggestion?

ivankravets, 2016-09-12:
Tests are located in the framework. See how we test it with Travis https://github.com/platformio/builder-framework-mbed/blob/develop/.travis.yml

zgoda, 2016-09-15:
Could you please remind me how do I build framework-mbed with new mbed version? I really can't find it now. Branch develop installs 124, branch with new builder installs 123 and I can't see any new branch that could install 125.

ivankravets, 2016-09-15:
@zgoda 

1. Go to `~/.platformio/packages/framework-mbed`
2. Backup `package.json` file and `platformio` directory
3. Remove all files form this dir
4. Clone here new mbed OS
5. Revert back files and dirs from #2 step
6. Update variants DB. See `~/.platformio/packages/framework-mbed/platformio/README.txt`

Thanks!

evark, 2016-09-16:
Version 126 just released
https://github.com/ARMmbed/mbed-os/tree/mbed_lib_rev126

ivankravets, 2016-09-17:
I updated mbed OS to 5.1.4/r126. However, I test is broken:
- https://travis-ci.org/platformio/platform-ststm32

@zgoda Is it possible that new mbed OS requires more resources? Also, do you have any ideas as for
* https://travis-ci.org/platformio/platform-nxplpc/builds/160646433
* https://travis-ci.org/platformio/builder-framework-mbed

Or something is wrong with our builder?

Thanks!

evark, 2016-09-18:
With r126 something is broken:

Compiling .pioenvs/nucleo_f103rb/FrameworkMbed-TARGET_STM32F1-99b21/stm32f1xx_hal.o
arm-none-eabi-gcc: error: mbed_config.h: No such file or directory

ivankravets, 2016-09-18:
The latest ARM mbed has broken project generator. We use it to fetch build information (flags, etc). I'll fix it manually and open issue in their repository.

evark, 2016-09-18:
This file mbed_config.h is something new. It is dynamically generated by mbed building system, I didn't tried to build my sources with mbed actually.

ivankravets, 2016-09-18:
1. Issue is opened https://github.com/ARMmbed/mbed-os/issues/2736
2. I've jsut repacked package. Try `pio update` and buidl again.

Does it work?

ivankravets, 2016-09-18:
Don't forget to install development version of platform:
```
pio platform uninstall ststm32
pio platform install https://github.com/platformio/platform-ststm32.git
```

evark, 2016-09-18:
Building is working again.  I didn't test the binaries on targets but seems OK

ivankravets, 2016-09-18:
Thanks! I'll release it this week.

zgoda, 2016-09-19:
@ivankravets yes, I see newer versions require more and more, starting with rev. 123 (basic blinky on bluepill compiles to almost 57 KB firmware compared to ~30 KB on disco-f051) so I suppose something does not work quite right.

I'm not able to build anything on bluepill now with mbed 124:

```
Linking .pioenvs/bluepill_f103c8/firmware.elf
.pioenvs/bluepill_f103c8/FrameworkMbedHalCommon/retarget.o: In function `__wrap_main':
retarget.cpp:(.text.__wrap_main+0xe): undefined reference to `main'
collect2: error: ld returned 1 exit status
*** [.pioenvs/bluepill_f103c8/firmware.elf] Error 1
```

ivankravets, 2016-09-19:
@zgoda give me please an output of `pio platform show ststm32`

zgoda, 2016-09-20:
Sorry, disregard this, I had no `main()` in source, all is fine.

ivankravets, 2016-09-20:
Let's wait when they resolve this issue https://github.com/ARMmbed/mbed-os/issues/2736

ivankravets, 2016-10-08:
Released! Please run `pio update`

closed 2016-10-08 by ivankravets