platformio/platform-ststm32 #8: arduino-blink is not working

skabashnyuk, 2016-09-20:
```
[18:59:44]sj:arduino-blink[develop]#: pwd
/Users/sj/dev/src/thirdparty/platform-ststm32/examples/arduino-blink
[18:59:45]sj:arduino-blink[develop]#: pio run
[Tue Sep 20 18:59:50 2016] Processing maple (platform: ststm32, board: maple, framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
AttributeError: 'PlatformBoardConfig' object has no attribute 'id':
File "/usr/local/Cellar/platformio/3.0.1/libexec/lib/python2.7/site-packages/platformio/builder/main.py", line 134:
env.SConscript("$BUILD_SCRIPT")
File "/Users/sj/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 551:
return _SConscript(self.fs, *files, **subst_kw)
File "/Users/sj/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 260:
exec _file_ in call_stack[-1].globals
File "/Users/sj/.platformio/platforms/ststm32/builder/main.py", line 177:
target_elf = env.BuildProgram()
File "/Users/sj/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/usr/local/Cellar/platformio/3.0.1/libexec/lib/python2.7/site-packages/platformio/builder/tools/platformio.py", line 56:
f.lower().strip() for f in env['PIOFRAMEWORK'].split(",")
File "/Users/sj/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/usr/local/Cellar/platformio/3.0.1/libexec/lib/python2.7/site-packages/platformio/builder/tools/platformio.py", line 251:
SConscript(env.GetFrameworkScript(f))
File "/Users/sj/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 614:
return method(*args, **kw)
File "/Users/sj/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 551:
return _SConscript(self.fs, *files, **subst_kw)
File "/Users/sj/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 260:
exec _file_ in call_stack[-1].globals
File "/Users/sj/.platformio/platforms/ststm32/builder/frameworks/arduino.py", line 33:
if board.id == "bluepill_f103c8":
======================================================================================== [ERROR] Took 0.37 seconds ========================================================================================

[Tue Sep 20 18:59:51 2016] Processing genericSTM32F103RB (platform: ststm32, board: genericSTM32F103RB, framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
AttributeError: 'PlatformBoardConfig' object has no attribute 'id':
File "/usr/local/Cellar/platformio/3.0.1/libexec/lib/python2.7/site-packages/platformio/builder/main.py", line 134:
env.SConscript("$BUILD_SCRIPT")
File "/Users/sj/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 551:
return _SConscript(self.fs, *files, **subst_kw)
File "/Users/sj/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 260:
exec _file_ in call_stack[-1].globals
File "/Users/sj/.platformio/platforms/ststm32/builder/main.py", line 177:
target_elf = env.BuildProgram()
File "/Users/sj/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/usr/local/Cellar/platformio/3.0.1/libexec/lib/python2.7/site-packages/platformio/builder/tools/platformio.py", line 56:
f.lower().strip() for f in env['PIOFRAMEWORK'].split(",")
File "/Users/sj/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/usr/local/Cellar/platformio/3.0.1/libexec/lib/python2.7/site-packages/platformio/builder/tools/platformio.py", line 251:
SConscript(env.GetFrameworkScript(f))
File "/Users/sj/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 614:
return method(*args, **kw)
File "/Users/sj/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 551:
return _SConscript(self.fs, *files, **subst_kw)
File "/Users/sj/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 260:
exec _file_ in call_stack[-1].globals
File "/Users/sj/.platformio/platforms/ststm32/builder/frameworks/arduino.py", line 33:
if board.id == "bluepill_f103c8":

````
any ideas what is wrong?

ivankravets, 2016-09-20:
You need PlatformIO 3.1.0 where STM32Duino was introduced. See https://community.platformio.org/t/platformio-ide-1-6-0-and-platformio-cli-3-1-0/777

Try
```
brew update && brew upgrade
```

skabashnyuk, 2016-09-20:
Yes it works.
My bad. I thought I've updated brew recently and yesterday it was not working.
Solution is very simple thanks.

closed 2016-09-20 by skabashnyuk

ivankravets, 2016-09-20:
Really, it's our problem. We should not allow launching of development platform with PlatformIO Core that doesn't satisfy requirements. [We have these requirements](https://github.com/platformio/platform-ststm32/blob/develop/platform.json#L9) in manifest but PlatformIO Core doesn't use them. I'll implement it later. Thanks.