platformio/platform-ststm32 #10: Can't build for Blue Pill with Arduino framework

vshymanskyy, 2016-09-21:
Example of failed build task:
https://travis-ci.org/blynkkk/blynk-library/jobs/161392538

ivankravets, 2016-09-21:
@ubis, would be thankful for any help on it.

ubis, 2016-09-22:
Seems like **_sbrk** function is in variants/mcu_variant/syscalls.c.
For some reason it's not linking, due that undefined reference appears. 
I will look into this tomorrow, I'm really busy today, sorry.

ubis, 2016-09-23:
Okay, seems like I found what's causing this. I've added "libc" to builder LIBS(array..?) And it linked properly. However, I found that there are libgcc/libc/libm used in ld scripts, and as I noticed, builder script have:

`LIBS=["c", "gcc", "m", "stdc++", "nosys"]`

What's the difference? If I remove **c** from LIBS and add **libc** it won't link. But, if I add **libc** and keep **c**, it does link.

ivankravets, 2016-09-23:
Check linker output with `pio run --verbose`. Is this command correct?

ubis, 2016-09-23:
Okay, this is wierd. Without "libc":

> -lc -lgcc -lm -lstdc++ -lnosys

With "libc":

> -lc -lgcc -lm -lstdc++ -lnosys -lc 

So I changed "libc" to "c" and it still worked. I thought, maybe position could be wrong... But, it doesn't work if there is only one "c" in LIBS. Even this one works:

> -lgcc -lm -lc -lc

ivankravets, 2016-09-23:
1. `libc` == `c`. The same, use `c`.
2. Double `c` is very strange. Something is wrong with GCC LD circle finder. 

Could you try to replace `env.BuildLibrary` with `env.BuildSources`?
* https://github.com/platformio/platform-ststm32/blob/develop/builder/frameworks/arduino.py#L131
* https://github.com/platformio/platform-ststm32/blob/develop/builder/frameworks/arduino.py#L136

ubis, 2016-09-23:
So i've changed `env.BuildLibrary` with `env.BuildSources`.
Still no luck - won't link with one `c` and still does with double `c`.

ivankravets, 2016-09-23:
Ok, let's keep `env.BuildLibrary` and double `c`. I don't want to waste your time. We will take a look at it later.

ubis, 2016-09-23:
Ups, my bad sorry(again, lol). I actually changed it on my fork and I didn't moved it to stm platform folder.

Now, it produces error:

> *** [.pioenvs\genericSTM32F103RB\firmware.elf] Implicit dependency .pioenvs\genericSTM32F103RB\None not found, needed by target .pioenvs\genericSTM32F103RB\firmware.elf.

Since now it doesn't make *.a lib, it can't go further, I think..
Eh.. And that's okay. I'm free now, and actually we are wasting your time with this platform ;)

ivankravets, 2016-09-23:
Would be good to see build output from Make or Arduino IDE, to compare commands. How do you program STM32Duino without PlatformIO?

ubis, 2016-09-23:
You can use Arduino IDE, and get arduino stm32 from [repo](https://github.com/rogerclarkmelbourne/Arduino_STM32). It has [installation page](https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Installation) aswell.

But, I could upload log to pastebin or so.

ivankravets, 2016-09-23:
Would be thankful for pastebin log.

ubis, 2016-09-23:
Here it is: [http://pastebin.com/8MnCWLYv](http://pastebin.com/8MnCWLYv). 
It's kinda ugly. And when I was adding support for STM32Duino, I had to compare aswell. It has some duplicate flags.

ivankravets, 2016-09-23:
> -lm -lgcc -mthumb -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -Wl,--warn-unresolved-symbols

I don't see here `c` lib. Maybe this is problem? Need to remove all libs and pass only `m` and `gcc`.
Within builder/framework/arduino.py 
```
env.Replace(LIBS=["m", "gcc"])
```

leodesigner, 2016-10-06:
Hi,
I have exactly the same problem.
At the moment I am fixing this with:
```
env.Replace(LIBS=["c", "gcc", "m", "stdc++", "nosys", "c"])
```
in arduino.py (line 124)

leodesigner, 2016-10-06:
Also - this may be relevant to compiler/linker options:
The binary file produced by the platformio is a bit bigger comparing to the file compiled by the stm32duino(Arduino 1.6.12). It contains debugging info probably - text like:
```
.platformio/packages/framework-arduinoststm32/STM32F1/cores/maple/HardwareTimer.cpp
```
in the resulting binary file.

vshymanskyy, 2016-10-11:
Any ideas how this can be fixed?

ivankravets, 2016-10-11:
Development platform will be released today.

commit 557f32e8 (2016-10-11): Fix linker error "undefined reference to _sbrk" for STM32Duino // Resolve #10
closed 2016-10-11 by ivankravets