platformio/platform-ststm32 #9: Simple upload test with maple board not working

albertoferna, 2016-09-21:
I have just switched to 3.1.0 and I can upload a simple sketch using the maple bootloader.
Looking at the command, and comparing it to a working version of the same I see a couple of differences.
Standard platformio install upload command for maple:
<some path>/maple_upload /dev/ttyACM0 2 0x1EAF:0x0003 <some path>/firmware.bin
working command:
<some path>/maple_upload ttyACM0 1 0x1EAF:0x0003 <some path>/firmware.bin
I can change 2 to 1 in main.py (usbid = '1 %s:%s' % (usbids[0][0], usbids[0][1])) but that seems hard coded and I don't know what effects that could have.
Also,  I can set upload_port = ttyACM0 in platformio.ini, but that seems like a work around as it would change depending of what has been plugged in the system.
Is there a better way to do it without breaking future updates?

ivankravets, 2016-09-21:
@ubis is this our issue?

ubis, 2016-09-21:
These '1' and '2' are actually revisions of dfu bootloader. Mostly all generic boards use newer ('2') bootloader so yeah, I kept it. However, some Maple boards still use an older bootloader ('1').
This could be changed by adding additional variable in platformio.ini (usb_bootloader_rev=1 / 2 or so..), but I'm sorry, I will be free only in upcoming weekend, so I could add this, later.

As for port, I think it should work with full path(with /dev/....), but just in case: could you try to change revision(that 2 to 1) and nothing else, also leave full upload port path and try to upload? If that won't work, /dev/... path could be removed before upload.

albertoferna, 2016-09-21:
That's in fact how I found that I needed to change that too. If I use the full path of the port it does not work. In my linux machine (ubuntu) it complains that it failed to open the serial device.
I'll read around to see if I can update the bootloader

albertoferna, 2016-09-21:
Sorry guys, wrong button. I did not mean to close the isse :(

ubis, 2016-09-23:
I've added ability to change Maple bootloader revision, see [Pull #12](https://github.com/platformio/platform-ststm32/pull/12).

As for path fix, i'm not sure how to do that. I can't see where I can change it, when port is automatically selected. And looking at bootloader script file for linux, it seems like it adds "/dev/" prefix for serial port. I could remove it there.

ivankravets, 2016-09-23:
@ubis @albertoferna , I'll fix that. Tell me please, we should use "port name" without full path for all OS?

ubis, 2016-09-23:
For Linux and OS X. Since Windows don't use /dev, yeah thats for all OS.

ivankravets, 2016-09-23:
Thanks for the info. I'll resolve this issue later

ivankravets, 2016-10-11:
I've just pushed new 1.3.0-alpha1 development version of this platform. Please re-test.

```
pio platform uninstall ststm32
pio platform install https://github.com/platformio/platform-ststm32/archive/develop.zip
```

CC @ubis, @albertoferna, @fabriciopk, @justgook, @vshymanskyy, @skabashnyuk

albertoferna, 2016-10-12:
I have run some test on that development version. I still have problem uploading. Right now, I have some boards with the original firmware (olimex) and some with the STM32Duino bootloader 1.0
I did a clean install of platformio plus the instructions above. If I try to flash a board that has revision 2 bootloader (STM32duino) with default platformio settings the board does not come back back to running. I have to manually set dfu mode and flash it from command line with bootloader revision 1. 
Strictly speaking, this does not seem to be a problem with platformio as it would happen the same if I use maple_upload from the command line with revision 2.
It I do the same with a board that has the original bootloader I get an error as it is to be expected.
I got it to work in a somehow hackish way by using upload_flags = 1 and some code in main.py (https://github.com/albertoferna/platform-ststm32/commit/767dd482ec617a18b8de6d9b809afa0073bbdfc4)
This way it works fine for both revision 1 and 2 of the firmware

ivankravets, 2016-10-12:
@albertoferna what is your board?

How about to have `upload_protocol = dfu_v1` and `upload_protocol = dfu_v2`options?

albertoferna, 2016-10-12:
@ivankravets my board is an Olimexino-STM32
That's perfectly fine with me. In fact, I tried to do something similar (dfu_1) but for some reason I could not get it to work

ivankravets, 2016-10-12:
@albertoferna Strange... Do you use `board = maple`? I've pushed new version of dev.paltform 1 hour ago. Could you repeat this step https://github.com/platformio/platform-ststm32/issues/9#issuecomment-253068416 ?

P.S: " I tried to do something similar (dfu_1) but for some reason I could not get it to work" - I've not implemented it yet. That was a proposition.

albertoferna, 2016-10-12:
I understood. I meant that I try to modify the code myself to work that way but I could not get it to work. I don't know enough of the ins and outs of the system

ivankravets, 2016-10-12:
I mean that you don't use the latest version of this dev platform. Please repeat these steps https://github.com/platformio/platform-ststm32/issues/9#issuecomment-253068416 where all should work from a box without any extra settings in `platformio.ini`. See my platformio.ini:

```
[env:maple]
platform = ststm32
board = maple
framework = arduino
```

Build output of `pio run -d /tmp/asd -v -t upload --upload-port /dev/null`:
```
Use manually specified: /dev/null
maple_upload null 1 1EAF:0003 /private/tmp/asd/.pioenvs/maple/firmware.bin
Failed to open serial device.
```

albertoferna, 2016-10-12:
With this version and firmware stm32duino (revision 2 as we have been calling it) the upload works nicely. 
With revision 1 (original firmware) it behaves a little more erratic. It fails some times but repeating the flashing seems to work

ivankravets, 2016-10-12:
May I close this issue and make an official release?

albertoferna, 2016-10-12:
Yes from my point of view regarding this issue.

justgook, 2016-10-12:
with last develop version 
```
⚡pio run --target upload -v
[Wed Oct 12 20:31:51 2016] Processing stm32f103c8t6 (upload_protocol: serial, platform: ststm32, board: bluepill_f103c8, framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------

Collected 25 compatible libraries
Looking for dependencies...
Project does not have dependencies
BeforeUpload(["upload"], [".pioenvs/stm32f103c8t6/firmware.bin"])
Auto-detected: /dev/cu.SLAB_USBtoUART
serial_upload cu.SLAB_USBtoUART {upload.altID} {upload.usbID} /Users/gook/Projects/arduino_ST32/.pioenvs/stm32f103c8t6/firmware.bin
Failed to init device.
stm32flash Arduino_STM32_0.9

http://github.com/rogerclarkmelbourne/arduino_stm32

Using Parser : Raw BINARY
Interface serial_posix: 230400 8E1

*** [upload] Error 1
```

ivankravets, 2016-10-12:
Could you try with Arduino IDE? Does it work?

justgook, 2016-10-13:
that works with:
`./stm32flash -w ~/Projects/arduino_STM32/.pioenvs/stm32f103c8t6/firmware.bin -v -g 0x0 /dev/cu.SLAB_USBtoUART`

ivankravets, 2016-10-13:
@ubis could you explain me why I don't see `stm32flash` option in Arduino IDE? Is this official way to upload firmware? In which cases we should use it?

@justgook could you try to flash with `upload_protocol = dfu`?

ubis, 2016-10-13:
@ivankravets `stm32flash ` is uploader binary. Arduino IDE uses script, called `serial_upload` or `serial_upload.bat`(for windows) which calls `stm32flash`. You could get more info in arduino_stm/STM32F1/platform.txt

ivankravets, 2016-10-13:
The source code of this script is `$(dirname $0)/stm32flash/stm32flash -g 0x8000000 -b 230400 -w "$4" /dev/"$1" `.

Where @justgook has used `./stm32flash -w ~/Projects/arduino_STM32/.pioenvs/stm32f103c8t6/firmware.bin -v -g 0x0 /dev/cu.SLAB_USBtoUART`

Sorry, I don't have this board and can't test it. Should be the one solution that works all time.

ubis, 2016-10-13:
Parameter `-g` means address, for serial bootloader, `0x8000000 ` is used, for dfu, somewhere it's `0x8005000 ` or `0x8002000`. So, that source code of script should be used, as it's used with Arduino IDE.

justgook, 2016-10-13:
so `upload_protocol = serial` success
how to burn with dfu, what an how must be connected - have no clue

ivankravets, 2016-10-13:
@justgook that is strange! `serial` protocol should be used by default. Could you share here `platformio.ini`?

justgook, 2016-10-13:
yup - serial is by default - and now it works.. (just downloaded last version via )
```
⚡pio platform uninstall ststm32
⚡pio platform install https://github.com/platformio/platform-ststm32/archive/develop.zip
 
```

justgook, 2016-10-13:
but have no clue how to setup env for `dfu`

ivankravets, 2016-10-13:
Please ignore these posts with `dfu`, if `serial` works, that is good.

closed 2016-10-13 by ivankravets

justgook, 2016-10-13:
waiting for official releas

ivankravets, 2016-10-13:
The platform has been released yesterday :)

Try `pio platform uninstall ststm32` and then run project. It should install the latest dev platform and packages.

commit 3f6dbd1b (2016-10-11): Refactor support for STM32Duino // Resolve #9