platformio/platform-ststm32 #11: Program linking working in previous version not working in 3.1

albertoferna, 2016-09-22:
I have a program that uses arduino's SD library. It used to compile and work (using 3.0 development version and std32duino) but it gives a linker error with the latest version. I reduced the code to it minimum:
`#define SD_CS 25
#include <Arduino.h>
#include <SPI.h>
#include <SD.h>

void setup() {
  SPI.setModule(2);
  pinMode(SD_CS, OUTPUT);
  SD.begin(SD_CS);
}

void loop() {
}`

I get the error: undefined reference to `_sbrk' when linking. I have tried to set lib_ldf_mode to both 1 and 2. I have also tried installing the SD library from within platformio IDE as well as add it to the libs folder of the project.
Any ideas why this is not working with this version?

ivankravets, 2016-09-22:
Duplicate https://github.com/platformio/platform-ststm32/issues/10

closed 2016-09-22 by ivankravets

albertoferna, 2016-09-22:
Upps, sorry for the duplicate, missed that