platformio/platform-ststm32 #13: Request: SPL support for stm32f103-series

darkfibre-nl, 2016-09-30:
I have recently acquired a few STM32F103C8T6 (Cortex-M3)  & STM32F030F4P6 (Cortex-M0) mini boards, would you mind adding CMSIS / SPL support for those to this platform?

I see the STM32L1, STM32F3, STMF4 series have, so I hope it isn't too much work.

Thanks!

ikiselev, 2018-04-01:
There are a lot of cheap boards based on STM32F103C8T6 (look-like arduino nano, etc) and examples (spl). It is very good to start learning with. But unfortunately, examples are useless without SPL.
Maybe, anything new going on? Or, how to integrate F103 with SPL on my own?

ikiselev, 2018-04-02:
Sorry, did it on my own. It's simpler, than I thought:
1. In genericSTM32F103C8.json change line to: "extra_flags": "-DSTM32F103x8 -DSTM32F10X_MD", add "spl" in "frameworks" node.
2. Download SPL library from st, unzip it to ~/.platformio/packages/framework-spl. Follow the structure (mkdir cmsis/variants/stm32f1). Startup file (startup_stm32f10x.c). Assembler file was not working

bobvanderlinden, 2018-12-01:
I ran into the same issue. What would be a good first step to address this issue? Is there some documentation on how to add frameworks/platforms?