platformio/platform-ststm32 #14: Problem to burn bluepill_f103c8 via UART (arduino)

justgook, 2016-10-09:
```
Looking for upload disk...
Auto-detected: /dev/cu.SLAB_USBtoUART
Uploading .pioenvs/stm32f103c8t6/firmware.bin
sh: ~/.platformio/packages/framework-arduinoststm32/tools/macosx/: is a directory
*** [upload] Error 126
```

But then i do same with CLI:
```
~/.platformio/packages/framework-arduinoststm32/tools/macosx/stm32flash:$ ./stm32flash -w ~/Projects/test_ST32/.pioenvs/stm32f103c8t6/firmware.bin -v -g 0x0 /dev/cu.SLAB_USBtoUART
```

and that burns normally

for mbed - for now no success at all..

ivankravets, 2016-10-09:
Could you run it with `--verbose` option?

justgook, 2016-10-10:
today later - when be at home

zgoda, 2016-10-10:
To be precise - "bluepill_f103c8" is mbed platform and the only upload protocol that it supports now is stlink. I guess "stm32f103c8t6" is stm32duino platform.

justgook, 2016-10-10:
so for now there is no official support of bluepill_f103c8 with arduino (stm32duino):
but i can build project via:
```
[env:stm32f103c8t6]
platform = ststm32
framework = arduino
board = bluepill_f103c8
```
code:
```
#include "Arduino.h"
void setup() {
  pinMode(PC13, OUTPUT);
}

void loop() {
  digitalWrite(PC13, HIGH);
  delay(1000);
  digitalWrite(PC13, LOW);
  delay(1000);
}
```

justgook, 2016-10-10:
I have that - and im able to build app via platformio, and then burn it via CLI app and usb2UART addapter (by the way that was only way how "Disable the flash write-protection" that was set by default on that MCU) (stm32flash -w ~/Projects/test_ST32/.pioenvs/stm32f103c8t6/firmware.bin -v -g 0x0 /dev/cu.SLAB_USBtoUART)  
[image]

zgoda, 2016-10-10:
It is supported in both mbed and stm32duino. Please disregard my comment, I think I messed up stm32duino board name with mbed target name.

justgook, 2016-10-10:
```
gook@work:~/Projects/arduino_ST32⚡pio run -v --target upload
[Mon Oct 10 21:16:09 2016] Processing stm32f103c8t6 (platform: ststm32, board: bluepill_f103c8, framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------
Collected 24 compatible libraries
Looking for dependencies...
Project does not have dependencies
MethodWrapper(["upload"], [".pioenvs/stm32f103c8t6/firmware.bin"])
Auto-detected: /dev/cu.SLAB_USBtoUART
/Users/gook/.platformio/packages/framework-arduinoststm32/tools/macosx/ /dev/cu.SLAB_USBtoUART /Users/gook/Projects/arduino_ST32/.pioenvs/stm32f1
03c8t6/firmware.bin
sh: /Users/gook/.platformio/packages/framework-arduinoststm32/tools/macosx/: is a directory
*** [upload] Error 126
```

ivankravets, 2016-10-10:
@justgook Could you try `board = stm32f103c8t6` instead `board = bluepill_f103c8`?

justgook, 2016-10-10:
Error: Unknown board ID 'stm32f103c8t6'

ivankravets, 2016-10-10:
@justgook please sorry, it should be `board = genericSTM32F103C8`

justgook, 2016-10-10:
Totally same result:
```
gook@work:~/Projects/arduino_ST32⚡pio run -v --target upload
[Tue Oct 11 00:16:14 2016] Processing stm32f103c8t6 (platform: ststm32, board: genericSTM32F103C8, framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------
Collected 24 compatible libraries
Looking for dependencies...
Project does not have dependencies
MethodWrapper(["upload"], [".pioenvs/stm32f103c8t6/firmware.bin"])
Auto-detected: /dev/cu.SLAB_USBtoUART
/Users/gook/.platformio/packages/framework-arduinoststm32/tools/macosx/ /dev/cu.SLAB_USBtoUART /Users/gook/Projects/arduino_ST32/.pioenvs
/stm32f103c8t6/firmware.bin
sh: /Users/gook/.platformio/packages/framework-arduinoststm32/tools/macosx/: is a directory
```

ubis, 2016-10-11:
You need to set upload protocol in platformio ini. For example, if you want to use STM32 serial, you need to write this line into platformio.ini:

`upload_protocol = serial`
or if using dfu:
`upload_protocol = dfu`

justgook, 2016-10-11:
btw is there somewhere DFU manual - how to enable it and how to use it ?

ubis, 2016-10-11:
You just need to set upload protocol as dfu and it should be fine.
Although there are some issues with older dfu/maple revisions and we didn't solved it yet.

justgook, 2016-10-11:
need to check later at home..

fabriciopk, 2016-10-11:
I upload the .bin(.pioenvs/stm32f103c8t6/firmware.bin) using openocd and stlink and now I can't re-flash the board. The swd is disable during the build?

ivankravets, 2016-10-11:
@ubis that is not good workflow. Some protocol should be set by default when user didn't specify it.
Which protocol we can use by default?

ubis, 2016-10-11:
Yeah, actually I had "serial" protocol as default back then, but I removed it when it was merged into this repo. So yeah, "serial" protocol could be used by default.

ivankravets, 2016-10-11:
Thanks, I'll fix it.

I'm just working on refactoring the support for Arduino/stm32. Need to move all tools from framework package to own OS related packages where our package manager will automatically handle them.

I'll report here about results.

ivankravets, 2016-10-11:
Development platform will be released today.

closed 2016-10-11 by ivankravets

justgook, 2016-10-11:
just downloaded `develop` branch and got error (but looks good)

```
pio run -v --target upload
[Tue Oct 11 23:10:12 2016] Processing stm32f103c8t6 (platform: ststm32, board: genericSTM32F103C8, framework: arduino)

-------------------------------------------------------------------------------------------------------------------------------------------------
Collected 25 compatible libraries
Looking for dependencies...
Project does not have dependencies
MethodWrapper(["upload"], [".pioenvs/stm32f103c8t6/firmware.bin"])
Auto-detected: /dev/cu.SLAB_USBtoUART
/Users/gook/.platformio/packages/framework-arduinoststm32/tools/macosx/serial_upload /dev/cu.SLAB_USBtoUART {upload.altID} {upload.usbID} /Users/
gook/Projects/arduino_ST32/.pioenvs/stm32f103c8t6/firmware.bin /Users/gook/Projects/arduino_ST32/.pioenvs/stm32f103c8t6/firmware.bin
Cannot handle device "/dev//dev/cu.SLAB_USBtoUART"
stm32flash Arduino_STM32_0.9

http://github.com/rogerclarkmelbourne/arduino_stm32

Using Parser : Raw BINARY

Failed to open port: /dev//dev/cu.SLAB_USBtoUART
*** [upload] Error 1
```

ivankravets, 2016-10-11:
@justgook this issue is related with https://github.com/platformio/platform-ststm32/issues/9

ivankravets, 2016-10-12:
@justgook  could you re-test the latest dev version? https://github.com/platformio/platform-ststm32/issues/9#issuecomment-253068416

justgook, 2016-10-12:
im at work - i can test it when ill get home

ivankravets, 2016-10-12:
PIO Remote will resolve this issue and will allow to work with home's hardware from a work 😊

commit 012d7eb9 (2016-10-11): Use "serial" protocol by default for STM32Duino-based boards // Resolve #14