platformio/platform-ststm32 #15: bluepill_f103c8 cannot buil mbed-rtos without specific build_flags

justgook, 2016-10-10:
if not set any build flag RTOS fail with error:
`~/.platformio/packages/framework-mbed/rtos/rtx/TARGET_CORTEX_M/RTX_CM_lib.h:609:2: error: #error "no target defined"`

by looking what is inside that file i found closest `TARGET_STM32F103RB` - so need to add `build_flags = -D TARGET_STM32F103RB` in `platformio.ini` to be to be able build.

zgoda, 2016-10-10:
Until we get official mbed-rtos porting guide consider this board is not rtos capable (only mbed-lib or "mbed classic"). I've been told earlier this year it has not enough flash memory to support rtos but things may have changed since then. Anyway, I will not begin porting rtos until porting guide is released.

justgook, 2016-10-10:
so - today i try few more coding with rtos - and i got why it is totally useless on BluePill - after creating simple (REALLY SIMPLE) application that runs 2 leds and send some message to serial - i went out of memory - so do some RTOS (mbed-rtos) on bluepill is totally useless.
so my suggestion is close that ticket and add some info in documentation - "RTOS is possible run on BluePill, but You can run only Blink app, where 99% (of total MCU flash ) took mbed-rots  + 2 threads"

zgoda, 2016-10-11:
After merging mbed "classic" (or "mbed-lib") with mbed-os much of bloat has been introduced to the point that boards with less than 128KB flash are almost unusable, even if none of rtos features are used. Simple program that blinks onboard led and doing any communication over serial compiles to ~55KB firmware. I see no reasonable way out of this situation, the only solution would be to fork mbed classic from just before merge and maintain this unofficial fork, but considering the size of the framework it is not a "reasonable solution". Or is it?

justgook, 2016-10-11:
yup - totally agree - it totally useless - maybe it is time to look to some other Framework / RTOS. For such modules.. as I see there was some tickets (issues) about [ChibiOS/RT, FreeRTOS & NuttX](https://github.com/platformio/platformio/issues/160) so would be nice to try something like that also for those uder128KB STM chips.. (i could try, but cannot find starting point, on how create new framework / lib )

Testato, 2016-11-26:
first tip: 
bluepill use F103c8, the datasheet tell that is have only 64KB but in reality it ave 128KB :-)
but st-link open source version do not add it officlially, so platformio, that use st-link open source version, cannot upload abobe 64KB
But if you compile by platformio, and after manually use the firmware.bin by st-link official utility (windows program), you can normally upload your code
https://github.com/texane/stlink/issues/516

second tip:
mbed-os is not the only official version of mbed. Is there the mbed-classic also, it is officially mantained, and it do not include the rtos. So if you need rtos you can use mbed-os or mbed-classic+rtos.h
but if you do not need rtos, by using the mbed-classic you will create a very small .bin firmware.
This is because on mbed-os the main() itself is a thread, so you always use rtos also on a blink.

Platformio need two different framework for give to the user a very mbed experience.
https://community.platformio.org/t/mbed-double-framework/962

Testato, 2016-11-26:
>  the only solution would be to fork mbed classic from just before merge and maintain this unofficial fork
mbed-classic is not "merged" inside mbed-os. It is added to mbed-os, but it is a separate project.
mbed-classic is not present on github, only mbed-os (that include mbed-classic) is on github

the official reposotory for mbed-classic is this, and it is update 2day ago to ver 130
https://developer.mbed.org/users/mbed_official/code/mbed/shortlog

An official platformio framewrok for mbed-classic need use this repo, not the mbed-classic piece included on mbed-os