platformio/platform-ststm32 #16: Upload using openocd and stlinkv2

fabriciopk, 2016-10-11:
I upload the .bin(.pioenvs/stm32f103c8t6/firmware.bin) using openocd and stlink and now I can't re-flash the board. The swd is disable during the build?

ivankravets, 2016-10-12:
@ubis, please help us.

ubis, 2016-10-13:
I'm not really familiar with SWD and stuff, maybe you used some different flags?
Could you try to reflash firmware with ROM bootloader?

fabriciopk, 2016-10-14:
I upload the code using openocd with the same script that I use for others binaries, and it always work. But when I flash with the binaries that I build using platformio, the stlink not recognize the microcontroler any more. 
It's possible to use platformio to upload the binaries using stlink? I was reading about adding in the platformio.ini

UPLOAD_PROTOCOL = stlink

ivankravets, 2016-10-14:
@fabriciopk please re-test the latest dev build:
```
pio platform uninstall ststm32
pio platform install https://github.com/platformio/platform-ststm32/archive/develop.zip
```

Does it work?

ivankravets, 2016-10-17:
@fabriciopk do you have any news?

ivankravets, 2016-11-18:
Please reopen if you still have problems

commit e7ee381f (2016-10-14): Allow using stlink for stm32duino // Resolve #16
closed 2016-11-18 by ivankravets