platformio/platform-ststm32 #18: [Issue] Programming MultiTech mDot using PlatformIO instead of ARM mbed online compiler

JekoTronik, 2016-10-12:
Hello PlatformIO !

I am using the MultiTech mDot in my projects. Actually, I use mbed online compiler and MultiTech MDK to program the board.

I would like to do all of it using Platformio and a cable USB -> Serial (FTDI chip).

As the first step, I use a basic code :

```
#include "mbed.h"

int main() {
// pin PA_1 is connected to my LED
DigitalOut led(PA_1);

while (true) {
        led = !led;
        wait_ms(3000);
    }
}
```

It works fine when compiled with mbed online compiler and then uploaded firmware.bin with MultiTech MDK. But when I compile it with Platformio and upload the firmware.bin with Multitech MDK, it doesn't blink my LED.

Here is my .ini content :

> [env:mts_mdot_f411re]
> platform = ststm32
> board = mts_mdot_f411re
> framework = mbed

My setup :

- Windows 10
- PlatformIO IDE 1.6.0 | CLI 3.1.0
- Does someone have any clue ?

If you need any more information, please ask me !

Thank you in advance.

ivankravets, 2016-10-12:
@valeros do you have any ideas?

valeros, 2016-10-12:
Hi @JekoTronik !
Could you try to export your blink example as a Makefile project (GCC ARM Embedded) from the mbed online compiler and build it locally? Does it work as expected?

JekoTronik, 2016-10-19:
Hello @ivankravets and @valeros,
I am sorry to tell you that I don't have the device at my disposal anymore to continue the debugging.
Sorry about that.
Cheers and thank you taking time to help me.

frankleonrose, 2017-07-05:
(I'm interested in getting this working, but haven't had any success, yet. But I'll tell you what I tried.)

The same is true with the mbed.com IDE - no GCC export. However, it does support exporting as a Keil project. I don't know Keil, but the project file is XML and I was able to try some things.

First, I noticed that the exported `mbed_config.h` file had a bunch more settings than `~/.platformio/packages/framework-mbed/platformio/variants/MTS_MDOT_F411RE/mbed_config.h` (which was structurally right but essentially empty).

```
> #define MBED_CONF_DRIVERS_UART_SERIAL_RXBUF_SIZE       256    // set by library:drivers
> #define MBED_CONF_DRIVERS_UART_SERIAL_TXBUF_SIZE       256    // set by library:drivers
> #define MBED_CONF_NSAPI_PRESENT                        0      // FLR set by library:nsapi
> #define MBED_CONF_FILESYSTEM_PRESENT                   0      // FLR set by library:filesystem
> #define MBED_CONF_PPP_CELL_IFACE_BAUD_RATE             115200 // set by library:ppp-cell-iface
> #define MBED_CONF_PLATFORM_STDIO_CONVERT_NEWLINES      0      // set by library:platform
> #define MBED_CONF_PPP_CELL_IFACE_APN_LOOKUP            0      // set by library:ppp-cell-iface
> #define MBED_CONF_EVENTS_PRESENT                       0      // FLR set by library:events
> #define MBED_CONF_RTOS_PRESENT                         1      // set by library:rtos
> #define MBED_CONF_PPP_CELL_IFACE_AT_PARSER_BUFFER_SIZE 256    // set by library:ppp-cell-iface
> #define MBED_CONF_PLATFORM_STDIO_BAUD_RATE             9600   // set by library:platform
> #define MBED_CONF_PLATFORM_DEFAULT_SERIAL_BAUD_RATE    9600   // set by library:platform
> #define MBED_CONF_PPP_CELL_IFACE_AT_PARSER_TIMEOUT     8000   // set by library:ppp-cell-iface
> #define MBED_CONF_PLATFORM_STDIO_FLUSH_AT_EXIT         1      // set by library:platform
> // Macros
> #define UNITY_INCLUDE_CONFIG_H                                // defined by library:utest
```

Overwriting the framework version with the exported file, `pio run` didn't succeed until I disabled `MBED_CONF_NSAPI_PRESENT`, `MBED_CONF_FILESYSTEM_PRESENT`, and `MBED_CONF_EVENTS_PRESENT` and added `-I/Users/frank/.platformio/packages/framework-mbed/rtos/rtx/TARGET_CORTEX_M/` to my build flags.

However, still no luck. The .bin generated does not run on the mDot.

frankleonrose, 2017-07-05:
The next information I tried to extract from the Keil project file was the preprocessor definitions. They are: `-I/Users/frank/.platformio/packages/framework-mbed/rtos/rtx/TARGET_CORTEX_M/ -DTARGET_STM -DDEVICE_RTC=1 -DDEVICE_SLEEP=1 -DTOOLCHAIN_object -DUSE_PLL_HSE_EXTC=0 -DTARGET_STM32F4 -D__ASSERT_MSG  -DDEVICE_PORTINOUT=1 -DDEVICE_PWMOUT=1 -DTARGET_MTS_MDOT_F411RE -c -DTARGET_RTOS_M4_M7 -DDEVICE_SPISLAVE=1  -DDEVICE_PORTOUT=1 -DDEVICE_STDIO_MESSAGES=1 -DTARGET_RELEASE  -DDEVICE_SERIAL=1  -DDEVICE_ANALOGIN=1 -DTARGET_LIKE_CORTEX_M4 -D__CORTEX_M4 -DTARGET_CORTEX_M -DARM_MATH_CM4 -DDEVICE_I2C=1 -DDEVICE_PORTIN=1  -DHSE_VALUE=26000000 -DDEVICE_INTERRUPTIN=1 -DTARGET_STM32F411RE  -DTARGET_UVISOR_UNSUPPORTED -DVECT_TAB_OFFSET=0x00010000 -DTARGET_LIKE_MBED  -DDEVICE_SPI=1 -D__MBED__=1 -DTARGET_M4  -D__FPU_PRESENT=1 -DDEVICE_I2CSLAVE=1 -D__CMSIS_RTOS -DDEVICE_SPI_ASYNCH=1  -DTRANSACTION_QUEUE_SIZE_SPI=2 -D__MBED_CMSIS_RTOS_CM`

I had to remove `-DTOOLCHAIN_ARM_STD` to get a successful build.

Also, there were Keil options that were rejected by gcc: 
 - --preinclude=mbed_config.h
 - --split_sections
 - --no_rtti
 - --cpu=Cortex-M4.fp
 - --no_depend_system_headers
 - --md
 - --gnu -DTOOLCHAIN_ARM
 - --apcs=interwork

I have not yet translated those into equivalent gcc options. 

Successfully built .bin file, but again, no indication that it is running. 

Next step is to compare the structure of .bin files that work with the ones that don't.

frankleonrose, 2017-07-05:
Added GCC equivalent of Keil arguments
```
-include mbed_config.h 
-ffunction-sections 
-fno-rtti
-mcpu=cortex-m4 
-march=armv7e-m 
-mthumb-interwork 
```
to no avail. (Omitted -DTOOLCHAIN_ARM because triggered #include of unknown `rt_sys.h` and `--md` and `--no_depend_system_headers` are for generating make dependency lists.)

Also noted that startup assembly code and linker scripts for both builds are identical.

Next step is to connect with debugger and see where the PlatformIO code is breaking on the mDot.

frankleonrose, 2017-07-07:
With only
```
-mthumb-interwork
-Wl,--wrap,_memalign_r
```
in `build_flags` and (ultimately)
```
#define MBED_CONF_FILESYSTEM_PRESENT                   0
```
in the config file, I was able to use the PlatformIO debugger on the resulting code. I can also run gdb remote from the command line and load the .elf file and it will run.

However, none of the .bin files yet created by PlatformIO will run when copied over to the MULTITECH drive, the standard flashing method used with mbed.

ivankravets, 2018-12-11:
Could you reproduce this issue with the latest dev/platform version and mbed?