platformio/platform-ststm32 #19: Support for mbed 3.127

douardda, 2016-10-17:
When will mbed 3.127 be added? 

BTW, is there a way to install (locally) a yet unsupported version of mbed somehow in platformio?

Thanks,

David

douardda, 2016-10-17:
I'm also wondering if there is a way to use the future mbed-os 5.2 in platformio? I guess  I need to be able to somehow compile my own framework-mbed...

David

BTW platformio is really awesome, thanks

ivankravets, 2016-10-17:
> BTW, is there a way to install (locally) a yet unsupported version of mbed somehow in platformio?

Please go to `/.platformio/packages/framework-mbed` and unpack ARM mbed source files here. Keep `platformio` folder and `package.json` without changes.

douardda, 2016-10-18:
Thanks @ivankravets it works with mbed 3.127. 

I also gave a try to mbed-os-5.2 rc3 (which is my true goal here: I want to use the async api that is now available on all nucleo boards, especially the L432KC I'm using for my current project) and the layout of the repository have changed a lot, so the  platforms/ststm32/builder/frameworks/mbed/mbed.py file needs to be adapted. Have anyone already started to work on supporting mbed-os 5.2 ?

David

douardda, 2016-10-19:
I've been able to compile and upload some examples on a nucleo board using current mbed head (ie. something like mbed-os 5.2 rc3).  The serial examples is unstable (the device hang if I send chars too quickly) but that's probably an mbed-only pb. 

I'll try to push my code, but I'm pretty sure it's not correct.

douardda, 2016-10-19:
Ok I've pushed my code on https://github.com/douardda/builder-framework-mbed/tree/mbed-5.2

Note sure I should make a PR with this, more polishing might be necessary.

ivankravets, 2016-10-19:
Thanks a lot for your PR. Please keep it alive. We will back to it after 5.2 release.

douardda, 2016-10-19:
Ok so let me create a proper PR with my branch even if it will most probably require some more work.

douardda, 2016-10-20:
Hi, one more question: what is the code responsible for generating the eix files? They look generated from (at least) targets.json, and I'd like to update them for mbed 5.2, and there seems to be a small bug making flags duplicated several times).

ivankravets, 2016-10-20:
@douardda Please check `~/.platformio/packages/framework-mbed/platformio/README.txt`

douardda, 2016-10-21:
Sorry I missed this README file... thanks @ivankravets!

ivankravets, 2016-11-18:
Resolved in https://github.com/platformio/platform-ststm32/commit/40dc0b3e970b9a5279db567e6ebe8e135c748b3d

closed 2016-11-18 by ivankravets