platformio/platform-ststm32 #22: Support for maple mini

maxpowel, 2016-11-09:
Hi, I'm wondering if it's viable to add mapple mini to the supported devices list. Using the arduino ide and arduino stm32 it's possible but I don't know how hard would be to do the same with platformio.

Thank you

ivankravets, 2016-11-18:
@ubis what should be done on our side?

Thanks!

ubis, 2016-11-18:
I did a quick look at: [https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/boards.txt](https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/boards.txt) and seems like that board config maple_mini.json should be created:

`{
  "build": {
    "core": "maple",
    "cpu": "cortex-m3",
    "extra_flags": "-DMCU_STM32F103CB",
    "f_cpu": "72000000L",
    "hwids": [
        ["0x1EAF", "0x0003"],
        ["0x1EAF", "0x0004"]
    ],
    "ldscript": "flash.ld",
    "mcu": "stm32f103cb_maple_mini",
    "variant": "maple_mini"
  },
  "frameworks": ["arduino"],
  "name": "Maple Mini (17k RAM, 108k Flash)",
  "upload": {
    "disable_flushing": false,
    "maximum_ram_size": 17408,
    "maximum_size": 110592,
    "protocol": "dfu",
    "boot_version": 1,
    "require_upload_port": true,
    "use_1200bps_touch": false,
    "wait_for_upload_port": false
  },
  "url": "http://www.leaflabs.com/maple/",
  "vendor": "Leaflabs"
}`

or [http://pastebin.com/mZRkRzPP](http://pastebin.com/mZRkRzPP) (if above code indentation will be messed up).

Would be nice, if @maxpowel add this and test if it works. Also, maple_mini as maple, can use bootloader 1.0 or 2.0. When I did a quick look in arduino.py, i've noticed 87 line:

`if "stm32f103rb_maple" in board.get("build.mcu", ""):`

This line is used to check if it's maple board, similar check should be done for maple_mini aswell.
Sorry, I can't do anything much more, i'm busy with studies & other stuff.

maxpowel, 2016-11-19:
@ubis that worked perfectly, thank you!
But as you said, there are bootloader v1 and v2. If you upload a sketch selecting the invalid bootloader you will have to use stlink and erase the flash.
You created the version for bootlader v1 (the original) but for version v2 the changes are just the name and version bootloader:

{
  "build": {
    "core": "maple",
    "cpu": "cortex-m3",
    "extra_flags": "-DMCU_STM32F103CB",
    "f_cpu": "72000000L",
    "hwids": [
        ["0x1EAF", "0x0003"],
        ["0x1EAF", "0x0004"]
    ],
    "ldscript": "flash.ld",
    "mcu": "stm32f103cb_maple_mini",
    "variant": "maple_mini"
  },
  "frameworks": ["arduino"],
  "name": "Maple Mini Bootloader 2 (20k RAM, 120k Flash)",
  "upload": {
    "disable_flushing": false,
    "maximum_ram_size": 17408,
    "maximum_size": 110592,
    "protocol": "dfu",
    "boot_version": 2,
    "require_upload_port": true,
    "use_1200bps_touch": false,
    "wait_for_upload_port": false
  },
  "url": "http://www.leaflabs.com/maple/",
  "vendor": "Leaflabs"
}

http://pastebin.com/4AqyhE4H

ivankravets, 2016-11-19:
Thanks @ubis and @maxpowel

closed 2016-11-19 by ivankravets

maxpowel, 2016-11-19:
Should I create a pull request with this changes?

ivankravets, 2016-11-19:
@maxpowel I've already added all confgis and updated buidl script. See https://github.com/platformio/platform-ststm32/commit/58858dacba8aed6721cd2e6ba6cafca494f2942e

commit 58858dac (2016-11-19): Add support for Maple Mini Original and Maple Mini Bootloader 2.0 // Resolve #22