platformio/platform-ststm32 #26: Support CMSIS for STM32F0

zserge, 2016-11-28:
It seems like `framework-cmsis` has variants for F1, F2, F3, F4, L0 and L1, but not for F0. I probably could copy CMSIS sources into my project source tree and go without any frameworks, but in this case all `*.S` boot files would be built instead of just one. So as a better solution I suggest to add ST's CMSIS implementation for F0 as well into PlatformIO framework package.

ivankravets, 2016-11-28:
Here is located a [package](https://bintray.com/platformio/dl-packages/framework-cmsis/1.40300.0#files) and [build script](https://github.com/platformio/platform-ststm32/blob/develop/builder/frameworks/cmsis.py).

Please modify it and attach here. I'll re-deploy to our CDNs.

zserge, 2016-11-29:
[framework-cmsis.tar.gz](https://github.com/platformio/platform-ststm32/files/617820/framework-cmsis.tar.gz)

Attached is a re-packed CMSIS package. I've only added the `stm32f0` variant. I took all the files from the most recent STMCube archive.

I don't think that build script should be modified at all.

I haven't got a chance to test it on a real hardware, but at least now my existing CMSIS code compiles successfully.

schwert, 2020-01-01:
Is there anything I can do to get this moving?

Schrolli91, 2020-04-29:
I also love to see this feature

valeros, 2020-06-26:
Resolved in https://github.com/platformio/platform-ststm32/commit/a1fa0e4a16480927ceccb700d4e8468d70f1ee28

closed 2020-06-26 by valeros