platformio/platform-ststm32 #28: Support for Arduino STM32L4

blackketter, 2017-01-12:
Please add support for arduino-STM32L4.

See: https://github.com/GrumpyOldPizza/arduino-STM32L4, which supports the Butterfly and Ladybug by @kriswiner & @GrumpyOldPizza

Thanks!

ivankravets, 2017-02-28:
@ubis Does STM32Duino support L4?

GrumpyOldPizza, 2017-02-28:
There is some half hearted support for the NUCLEO-L476 ... But I guess that
was not the original question.

- Thomas

ivankravets, 2017-02-28:
I think we can merge STM32Duino with @GrumpyOldPizza arduino-STM32L4 into the one build script https://github.com/platformio/platform-ststm32/blob/develop/builder/frameworks/arduino.py

@ubis WDYT? How is your free time, university? 😊

GrumpyOldPizza, 2017-02-28:
> builder/frameworks/arduino.py

I would not want to have this merged with STM32Duino ... There is a STM32L0
core coming down the pipe, and one for nRF52 ;-) Given that they'd share
all the same flavour, perhaps it's wise to separate this.

- Thomas

PS: I am not using PlatformIO, so I cannot speak to the implications.

ivankravets, 2017-02-28:
@GrumpyOldPizza under "merge", I mean only "build script" based on PlatformIO Build System API. The source code of the cores will not be merged.

GrumpyOldPizza, 2017-02-28:
Oki. If it helps any, the build is closer to ArdunioCore-samd. Only tricky
part is that it produces 2 output files. One for direct flashing, and one
for OTA.

- Thomas

ubis, 2017-03-01:
@ivankravets @GrumpyOldPizza I could take a look at it this weekend or better, next week.

ubis, 2017-03-03:
Okay, quick update. I've added all boards from @GrumpyOldPizza repo:
```
===================================================================== [SUMMARY] =====================================================================

[SUCCESS]Environment dragonfly-l476re    
[SUCCESS]Environment butterfly-l433cc    
[SUCCESS]Environment ladybug-l432kc      
[SUCCESS]Environment nucleo_l432kc       
[SUCCESS]Environment nucleo_l476rg       

============================================================ [SUCCESS] Took 8.10 seconds ============================================================
```

Though builds built with PlatformIO are slightly bigger, same as with STM32F103's:
```
arm-none-eabi-size -B -d .pioenvs\dragonfly-l476re\firmware.elf
text       data     bss     dec     hex filename
49640      2132    7120   58892    e60c .pioenvs\dragonfly-l476re\firmware.elf
```

`Sketch uses 32776 bytes (6%) of program storage space. Maximum is 507904 bytes.`
Now only a few issues are left to deal with... After that, I will push changes into my repo, so @GrumpyOldPizza , @blackketter and others could test it before merging it here.

@ivankravets where STM32L4 core sources will be? At the moment, i've moved them to arduinoststm32 framework folder, together with STM32F1/F2/F3.
Also, there are options like USB Type and DOSFS Type. Didn't touched them yet. They could be set in platformio.ini file via build flags I guess...

GrumpyOldPizza, 2017-03-03:
Ok, I am caught out at a bad point here. I was about to do some mayor
restructuring. Naming of the board as well as directory structure. The
current layout is not conforming to what Arduino wants to have the cores to
follow.

- Thomas

ivankravets, 2017-03-03:
@ubis Thanks a lot! 👍 

>  After that, I will push changes into my repo

Would be good, I'll review the code. Please ping me.

> where STM32L4 core sources will be? At the moment, i've moved them to arduinoststm32 framework folder

That is a right package.

> Also, there are options like USB Type and DOSFS Type. Didn't touched them yet. They could be set in platformio.ini file via build flags I guess...

Can we move them to Board's JSON manifest?

GrumpyOldPizza, 2017-03-03:
Ok, stupid big dummies question. Is there a way for me to integrate those
options and such into a JSON file that goes along with the github archive
on my side ?

- Thomas

ivankravets, 2017-03-03:
@GrumpyOldPizza Good idea.

@ubis How about to move your contribution to @GrumpyOldPizza repository? Then just "include" main build file. I've done the same for ESP32 For Arduino, now they support integration with PlatformIO by self. See

* https://github.com/espressif/arduino-esp32/blob/master/tools/platformio-build.py

Or, you can keep current work as you do, then I'll move "build logic" to @GrumpyOldPizza  repository.

WDYT?

ubis, 2017-03-03:
@ivankravets I think I will push to my repo, since I got some issues and been struggling for a while now...
When it will be fixed, you could move it then.

Davey3D, 2017-06-17:
I installed the repo linked above but still get: 
Error: This board doesn't support Arduino framework!

[env:nucleo_l432kc]
platform = ststm32
board = nucleo_l432kc
framework = Arduino

boristomas, 2018-08-31:
@Davey3D i have exactly the same issue, any luck with that?

kriswiner, 2018-08-31:
The L4 Arduino core does support two Nucleo boards: NUCLEO-L432KC and
NUCLEO-L476RG. It is intended to support development boards like this
<https://www.tindie.com/products/TleraCorp/ladybug-stm32l432-development-board/>
one sold on Tindie.

I am not sure what you are trying to do, but if you have the appropriate
hardware, the instructions for using it can be found here
<https://github.com/GrumpyOldPizza/arduino-STM32L4>.

If you have questions, please ask.

ivankravets, 2018-12-11:
We added support for official STM32 Core for Arduino. See supported boards https://github.com/stm32duino/Arduino_Core_STM32/blob/master/boards.txt

Please re-test with upstream version https://docs.platformio.org/en/latest/platforms/ststm32.html#upstream

closed 2018-12-11 by ivankravets

darshanmirajkar, 2019-08-07:
Hello, @ivankravets why still you guys didn't add <https://github.com/GrumpyOldPizza/arduino-STM32L4> in platformio??