platformio/platform-ststm32 #30: Firmware written in Flash but program is not working?

amodamatya, 2017-01-30:
Hello, i'm trying to upload my code to mts_dragonfly_f411re . Even after successful upload, the code does not work. I have used the simple code of led blink:  

#include "mbed.h"
DigitalOut myled(D3); 
int main()
{ 
while(1)
{ 
myled = 1;
wait(0.2);
myled = 0;
wait(0.2);
} 
}

 This works perfectly with bin file made from mbed-cli and mbed online compiler. How can I fix this problem?

Djgreene730, 2017-01-31:
Have you tried playing with the upload_port and upload_protocol settings in your platformio.ini file? What does your platformio.ini file currently look like?

amodamatya, 2017-01-31:
platformio.ini file:

[env:mts_dragonfly_f411re]
platform = ststm32
board = mts_dragonfly_f411re
framework = mbed
upload_protocol = stlink
upload_port = /dev/ttyACM0

darkomen, 2017-09-19:
I have the same problem with one **blue pill board.** I compile and upload one firmware with platform io, and the firmware compiles and uploads correctly but this, doesn't execute in the MCU. If I use the same program with coo cox Ide the firmware execute perfectly. When I compile in platform IO, I get the follows error:
**_Warning! Cannot find a linker script for the required board! Firmware will be linked with a default linker script!_**

But the LD script exits.

ivankravets, 2018-12-11:
Could you re-test it with the latest dev/platform and mbed version? Please open PIO IDE Terminal and type `pio update`. Does it work now?

closed 2019-02-21 by ivankravets