platformio/platform-ststm32 #31: Missing "tool-stm32duino" for ARMv7 Host

Enverex, 2017-02-10:
Trying to upload the code via "platformio run -t upload" but unfortunately it bails out with...

```
PackageManager: Installing tool-stm32duino @ ~1.0.0
Error: Could not find a version that satisfies the requirement '~1.0.0' for your system 'linux_armv7l'
```

Worth noting that stm32flash does exist for ARMv7 but I'm not 100% sure what tool-stm32duino entirely consists of.

In the meantime, I've used "stm32flash -b 1000000 -i -23,23 /dev/ttyAMA0 -w .pioenvs/stm32/firmware.bin" but that seems to end at "Wrote address 0x08020000 (47.46%) Done" (rather than 100%) but I'm not sure if that's actually an issue or not.

ubis, 2017-02-27:
What mcu do you use? As far as I know, default cmdline flashing with stm32flash is:
`stm32flash -g 0x8000000 -b 230400 -w ".pioenvs/stm32/firmware.bin" /dev/ttyAMA0`
Could you try it and tell if it gets stuck(doesn't reach 100%) and also if sketch is working?

As for tool-stm32 package.. @ivankravets needs get dfu-util, stlink and stm32flash binaries for ARMv7, so package for ARM could be created.

anweiss, 2017-06-07:
Any updates on this? Also running in to the same issue

valeros, 2017-06-09:
`tool-stm32duino` for ARM has been added to our packages.

closed 2017-06-09 by valeros