platformio/platform-ststm32 #33: `.text' will not fit in region `FLASH'

rodriguise, 2017-03-10:
platformio.ini:
```
[env:xdot_l151cc]
platform = ststm32
board = xdot_l151cc
framework = mbed
build_flags = -Wl,--gc-sections -L./lib -lxDot-GCC_ARM
extra_script = extra_script.py
```

extra_script.py:
```
Import('env')

env.Append(
  LINKFLAGS=[
      "-Wl,--gc-sections,--relax",
      "-Os"
  ]
)

env.Append(
  CCFLAGS=[
      "-g",
      "-fdata-sections",
      "-ffunction-sections",
      "-Os"
  ]
)
```

I am trying to link in the libxDot-GCC_ARM.a from the mBed compiler. My project obviously builds fine there. 

When I attempt to build using platformio 2.0beta I get something like this:

```
Archiving .pioenvs/xdot_l151cc/lib/libMTS-Utils.a
Archiving .pioenvs/xdot_l151cc/lib/libxDot-mbed5.a
Indexing .pioenvs/xdot_l151cc/lib/libMTS-Utils.a
Indexing .pioenvs/xdot_l151cc/lib/libxDot-mbed5.a
Archiving .pioenvs/xdot_l151cc/lib/librtos.a
Indexing .pioenvs/xdot_l151cc/lib/librtos.a
Linking .pioenvs/xdot_l151cc/firmware.elf
/Users/rrodriguez/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/
bin/ld: .pioenvs/xdot_l151cc/firmware.elf section `.text' will not fit in region `FLASH'
/Users/rrodriguez/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/
bin/ld: region `FLASH' overflowed by 38020 bytes
collect2: error: ld returned 1 exit status
*** [.pioenvs/xdot_l151cc/firmware.elf] Error 1
```

I have tried both GCC 4.8.4 and 4.9.1. It seems to not want to strip the imported lib. The files its linking are tiny, so I'm assuming its that:

```
Roberts-iMac:appa-mbed-fw rrodriguez$ ls -alh .pioenvs/xdot_l151cc/lib/
total 224
drwxr-xr-x   8 rrodriguez  staff   272B Mar 10 08:18 .
drwxr-xr-x  13 rrodriguez  staff   442B Mar 10 08:18 ..
drwxr-xr-x   4 rrodriguez  staff   136B Mar 10 08:18 ISL29011
-rw-r--r--   1 rrodriguez  staff   5.0K Mar 10 08:18 libISL29011.a
-rw-r--r--   1 rrodriguez  staff     8B Mar 10 08:18 libMTS-Utils.a
-rw-r--r--   1 rrodriguez  staff    96K Mar 10 08:18 librtos.a
-rw-r--r--   1 rrodriguez  staff     8B Mar 10 08:18 libxDot-mbed5.a
drwxr-xr-x  13 rrodriguez  staff   442B Mar 10 08:18 rtos
```

ivankravets, 2018-02-19:
Does it work with the latest upstream version?  http://docs.platformio.org/en/latest/platforms/ststm32.html#stable-and-upstream-versions

nachocarballeda, 2018-03-21:
Same problem over here...

matthieu-labas, 2019-02-18:
If nobody is willing to update that, can it be closed as "invalid"?

closed 2019-02-18 by ivankravets