platformio/platform-ststm32 #34: HAL support for STM32 MCUs

konradhalas, 2017-03-13:
HAL is an "official" way to develop software for STM32 MCUs. It's supported by STM32CubeMX tool which is also "official" tool from STMicroelectronics.

It would be really nice to have HAL support integrated with PlatformIO.

DavidPowell, 2017-04-23:
I have looked into this, and it should be fairly straightforward to write a python script which converts a CubeMX generated project into the appropriate file layout for PlatformIO. Similar scripts have already been written for other build tools which give some hints: [a Makefile solution](https://github.com/baoshi/CubeMX2Makefile), and [one based on GNU Arm Eclipse](http://www.carminenoviello.com/2015/11/02/quickly-import-stm32cubemx-project-eclipse-project/).

Unfortunately my efforts in this direction hit a problem, in that I cannot specify sub-folders to include due to the folder names getting expanded incorrectly. This is important because I want to keep the HAL files in a separate folder to the real user code (and I don't think it makes sense for the HAL to be compiled as a separate library).

See the [discussion](https://community.platformio.org/t/include-folders-are-incorrectly-modified/1781?u=davidpowell). If there is some fix or workaround, then I should be able to finish off my hacked-together script. In the mean-time, my work-around is to use GNU Arm Eclipse.

valeros, 2017-06-12:
Hi @konradhalas , @DavidPowell !
We've added initial support for STM32Cube in the latest release and would be great if you could test it. Here is an [example](https://github.com/platformio/platform-ststm32/tree/develop/examples/stm32cube-hal-blink) for STM32Cube HAL.

konradhalas, 2017-06-12:
@valeros awesome! I will definitely check it out :)

DavidPowell, 2017-06-14:
@valeros Thanks for this

I upgraded my platform ststm32 to the latest git version, then run `pio update`. After that I was able to compile and upload the blinking example successfully to my nucleo_f103rb.

Unfortunately the example itself doesn't seem to work. Firing up gdb and openocd, it seems to be stuck in the HardFault_Handler().

valeros, 2017-06-14:
Hi @DavidPowell ! Thanks for the report!
Could you please try to compile firmware with this [linker script](https://drp.io/f/Hyb6lo0G-)? Just copy it to root dir of your project and add next line to your `platformio.ini`:
```
build_flags = -DF1 -Wl,-T"STM32F103XB_FLASH.ld"
```

DavidPowell, 2017-06-15:
Thanks @valeros, the blinking example now runs.

The timing is wrong, because `F_CPU=72000000L` is defined, whereas in the example the clock has not been configured, so it is using the default 8MHz internal oscillator. It doesn't seem possible to override this in the `build_flags` - the value appears twice and the compiler seems to take the first definition.

Also, using `build_unflags= -DF_CPU=72000000L` results in `TypeError: unhashable type: 'list':`, so I am unable to remove this default value.

For this HAL where all clock registers are typically modified manually, the best option could be to have no default value, and force the user to define the clock frequency manually.

valeros, 2017-06-19:
Hi @DavidPowell ! Many thanks for the report!
I think the timing is not wrong, as you can see it's just not configured, simply because we have one common example for all STM32 families, so it's user responsibility to properly configure clock settings for the board. Maybe we should split HAL examples for each STM32 family with clock initialization routine?

DavidPowell, 2017-06-19:
@valeros Of course the user should configure the clock registers according to their application. But there are many possible clock configurations which are equally valid. If I decide to run at 8 MHz, there should be a way for me to set F_CPU accordingly. PlatformIO currently assumes that my clock is 72MHz from the board definition, and I cannot figure out how to tell it that I am using a different value.

valeros, 2017-06-19:
@DavidPowell 
`F_CPU` is nothing more than a common macro for ST platform, it is not used in HAL framework, so I suppose HSI is used as a default clock source after reset. Anyway, in order not to confuse users, I've deleted `F_CPU` macro in HAL build script.
BTW as a user, would you like to have new-lib nano libs enabled by default or it's ok to enable them as needed?

DavidPowell, 2017-06-19:
@valeros Sorry, you are right, I guess I was confused by the `F_CPU` macro!

Yes the HSI is used after reset, but the HAL needs to be told this explicitly. Looking through STM's examples I see that it is necessary to call `HAL_RCC_ClockConfig` and `HAL_RCC_OscConfig` before `HAL_Delay` will work correctly. This involves creating `RCC_ClkInitTypeDef` and `RCC_OscInitTypeDef` structs.

Looking at the stm32f1xx HAL, I see that the definition of `RCC_OscInitTypeDef` is highly inconsistent even within the F1 line of CPUs, so I understand how it will be difficult to make a generic example across all the different CPU lines. 

Looking a bit closer, I notice that `HAL_RCC_GetSysClockFreq` (which is called by `HAL_RCC_ClockConfig` to set the timer frequency correctly) directly looks at the registers. So maybe it is possible to get accurate delays without calling `HAL_RCC_OscConfig`, but only call `HAL_RCC_ClockConfig` which may be more consistent across different CPUs. I will investigate this when I get a chance.

As for the libraries, I am not sure.  Honestly, I am still just playing around with the STM32 HAL, I have not used it very extensively, so I do not have an "expert" answer. Whatever you choose, I think the main priority would be for the end user to have full control. Perhaps an example showing how to include newlib-nano would be enough.

cblack3, 2017-07-25:
How is this going? Is there a temporary work-around?

DavidPowell, 2017-07-25:
@cblack3 The problem was only with examples, and more recently I tried them on a Nucleo F401RE and it worked fine. I'm not sure if it's the different board or the updates to platformio that made the difference (I'm using the git version).

In my real HAL based code (currently partly generated by CubeMX) the clock is properly configured anyway and the timing is correct.

ivankravets, 2017-11-10:
See

* https://github.com/platformio/platform-ststm32/tree/develop/examples/stm32cube-hal-blink
* https://github.com/platformio/platform-ststm32/tree/develop/examples/stm32cube-ll-blink

closed 2017-11-10 by ivankravets

vanbwodonk, 2019-04-09:
If someone confusing about  this(not accurate pll/ delay/ timing). Maybe your problem is HSE_VALUE. You need to defined it on `build_flags = -DHSE_VALUE=xxxx`. "xxxx" is your hardware Xtal.
I generate simple code with cubeMX and generated under Makefile Format. This is my platformio.ini
```
[platformio]
src_dir = Src
include_dir = Inc

[env:disco_f407vg]
platform = ststm32
board = disco_f407vg
framework = stm32cube
build_flags =
  -DHSE_VALUE=8000000
```
All work fine right now.
https://github.com/vanbwodonk/F407CubeMX_PlatformIO