platformio/platform-ststm32 #35: Build Error for nucleo_f429zi

eeyurdakul, 2017-03-21:
This version of ststm32 gives build error for nucleo_f429zi. 
1.7.0 version works ok, but when I upgrade it to 1.8.1, it started to give this error;

**Linking .pioenvs/nucleo_f429zi/firmware.elf
/Users/eeyurdakul/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/4.8.4/../../../../arm-none-eabi/bin/ld:
.pioenvs/nucleo_f429zi/STM32F429xI.ld.link_script.ld:90: syntax error
collect2: error: ld returned 1 exit status
*** [.pioenvs/nucleo_f429zi/firmware.elf] Error 1**

I tested blink example for all other boards, it works well with them, the problem occurs only for nucleo_f429zi.

valeros, 2017-03-22:
Hi @eeyurdakul !
Your problem might be caused by the old toolchain version in our packages.
Could you please try to update the toolchain manually? Just 6 small steps:
1. Open PlatfomIO home directory `/Users/eeyurdakul/.platformio/packages/`
2. Back up the current toolchain package (simply rename `toolchain-gccarmnoneeabi` to 
`z_toolchain-gccarmnoneeabi`)
3. Download the latest toolchain for your OS from [here](https://launchpad.net/gcc-arm-embedded/+download)
4. Create a new folder `toolchain-gccarmnoneeabi` and untar the new toolchain to this folder. 
5. Copy `package.json` file from `z_toolchain-gccarmnoneeabi` to the new folder `toolchain-gccarmnoneeabi`
6. Try to compile your firmware again.

tvr3000, 2017-04-28:
This worked for me, thanks

closed 2018-02-19 by ivankravets