platformio/platform-ststm32 #37: Maple Mini Upload

sstaub, 2017-03-25:
After uploading code to Maple Mini (Original), I have to do a reset and also to restart the serial terminal.
Can this done automatic?

stevstrong, 2017-04-16:
Which upload method do you use?

For the ST-Link probe, I think this is due to the used flasher utility.
Currently PIO only uses the "st-flash" utility which does not reset the board after uploading.
Arduino for STM32 (stm32duino) uses the ST link command line interface (ST-LINK_CLI.exe) to upload. This does reset the board after uploading.

So the solution for automatic reset would be to use the "ST-LINK_CLI.exe" instead of "st-flash" to upload.

sstaub, 2017-04-16:
It is a maple mini clone with an usb-port and it has no st-link adapter. It has it's own boot loader I think.

stevstrong, 2017-04-16:
In your first post you said that it is an MM original, now you say it is a clone.
Anyway, which bootloader is on board? Original or the [stm32duino bootloader](https://github.com/rogerclarkmelbourne/STM32duino-bootloader/tree/master/STM32F1/binaries)?

Which board definition file are you using?

sstaub, 2017-04-16:
The Maple Mini is a china clone with an original boot loader. Here the ini:
`[env:maple_mini_origin]
platform = ststm32
board = maple_mini_origin
framework = arduino`
I tried upload with st-link but I get an error.

stevstrong, 2017-04-16:
Have you tried to upload with Arduino IDE? Does it resets after upload?

Because it may happen that the original bootloader will not reset after upload.
The stm32duino bootloader 2.0 will do it for sure.

krish2487, 2017-04-16:
Maple does not use stlink for flashing.
It uses the dfu bootloader for flashing. Your code should have the USB serial connection active in your code and the appropriate bootloader flags to upload code. The device shows up as a virtual com port to upload the firmware

sstaub, 2017-04-16:
When uploading with the Arduino IDE, I need also do a reset, but I don't restart the serial monitor of the Arduino IDE. I tried to flash the new boot loader with the special upload sketch but with no luck.

stevstrong, 2017-04-16:
To get the new bootloader you have to follow [these steps](https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/stm32duino-bootloader).

sstaub, 2017-04-16:
I tried it with the Arduino IDE ans PlatformIO, but it fails every time.

stevstrong, 2017-04-16:
Then you have to triple-check all your steps.
Have you set BOOT1 pin to GND ? Can you enter the bootloader mode?
Please explain in more details what exactly does not work, paste hier the Arduino messages.

sstaub, 2017-04-16:
I used the uploader sketch described here [https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Bootloader](url)

stevstrong, 2017-04-16:
Please explain in more details what exactly does not work, paste hier the Arduino messages.

sstaub, 2017-04-16:
After uploading the sketch, I get this message on the serial monitor; After confirming I get the fail message.
[image]

stevstrong, 2017-04-17:
The device may have the flash protection on.
In this case you can use the Flash updater utility from ST.

sstaub, 2017-04-20:
I have uploaded the boot loader v2.0 via st-link. But the problem still exist, on PlatformIO also on Arduino IDE.

stevstrong, 2017-04-20:
Which OS?
On Windows it should work if you use the USb DFU upload method.

sstaub, 2017-04-20:
I'm on macOS Sierra.

stevstrong, 2017-04-20:
Check out [this forum post](http://www.stm32duino.com/viewtopic.php?f=3&t=1656&hilit=dfu+macos), maybe it helps.

ivankravets, 2018-02-19:
it should work with the latest upstream version. Please re-test

- http://docs.platformio.org/en/latest/platforms/ststm32.html#stable-and-upstream-versions

sstaub, 2018-02-20:
working with last version fails

[image]

that's the ini. file:
[env:maple_mini_b20]
platform = https://github.com/platformio/platform-ststm32.git
board = maple_mini_b20
framework = arduino

ivankravets, 2018-02-20:
What is your OS? Please try to restart VSCode.

sstaub, 2018-02-20:
Using macOS 10.3.3, VSCode Insider v1.21.0 Insider.
now getting following message:
[image]

ivankravets, 2018-02-20:
Could you provide a simple sketch to reproduce this issue? I tried https://github.com/platformio/platform-ststm32/tree/develop/examples/arduino-blink and it works.

If this a private project, you can send me@ikravets.com.

Thanks!

sstaub, 2018-02-20:
I figured out a problem with SPI functionally, it seems there is a bug in the library.
But now it is not possible to upload:
[image]

ivankravets, 2018-02-20:
Please open PIO IDE Terminal and type `pio run -v -t upload`. We need the full output of a build process to see command arguments.

sstaub, 2018-02-20:
stefans-mbp:blinky sstaub$ pio run -v -t upload
[Tue Feb 20 12:10:24 2018] Processing maple_mini_b20 (platform: https://github.com/platformio/platform-ststm32.git; board: maple_mini_b20; framework: arduino)
-------------------------------------------------------------------------------------------------
PLATFORM: ST STM32 > Maple Mini Bootloader 2.0
SYSTEM: STM32F103CBT6 72MHz 20KB RAM (120KB Flash)
DEBUG: CURRENT(blackmagic) EXTERNAL(blackmagic, jlink, stlink)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(light)
Collected 30 compatible libraries
Scanning dependencies...
No dependencies
arm-none-eabi-g++ -o .pioenvs/maple_mini_b20/firmware.elf -Os -Wl,--gc-sections,--relax -mthumb -mcpu=cortex-m3 -Wl,-T"flash.ld" .pioenvs/maple_mini_b20/src/main.cpp.o -L/Users/sstaub/.platformio/platforms/ststm32@src-15e0e87f41a00aa2f9602f71367ac29c/ldscripts -L/Users/sstaub/.platformio/packages/framework-arduinoststm32/STM32F1/variants/maple_mini/ld -L.pioenvs/maple_mini_b20 -Wl,--start-group .pioenvs/maple_mini_b20/libFrameworkArduinoVariant.a .pioenvs/maple_mini_b20/libFrameworkArduino.a -lgcc -lm -Wl,--end-group
Checking program size
text       data     bss     dec     hex filename
12968      2152     976   16096    3ee0 .pioenvs/maple_mini_b20/firmware.elf
Configuring upload protocol...
AVAILABLE: blackmagic, dfu, dfu, jlink, stlink
CURRENT: upload_protocol = dfu
MethodWrapper(["upload"], [".pioenvs/maple_mini_b20/firmware.bin"])
Auto-detected: /dev/cu.usbmodem1411
maple_upload /dev/cu.usbmodem1411 2 1EAF:0003 /Users/sstaub/Development/maple/blinky/.pioenvs/maple_mini_b20/firmware.bin
Failed to open serial device.
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
dfu-util: No DFU capable USB device available
Deducing device DFU version from functional descriptor length
*** [upload] Error 74
=================================== [ERROR] Took 1.74 seconds ===================================
stefans-mbp:blinky sstaub$

ivankravets, 2018-02-20:
@ubis do you have any ideas?

ubis, 2018-02-20:
Seeing previous messages, I suppose @sstaub is still using STM32 bootloader aka 2.0. Essentially maple mini 2.0 is same as generic STM32F103CB board.
Could you try to use `board = genericSTM32F103CB` and `upload_protocol = dfu`?

In order to prevent this, we could use F103CB config when this board is used. 
@ivankravets  should we use F103CB's config or keep like it is now?

sstaub, 2018-02-20:
Tried it, but also doesn't work.

sstaub, 2018-02-20:
I have the same problem with a bluepill board with Arduino boot loader(from Robotdyn).
http://robotdyn.com/stm32-arm-arduino-mini-system-dev-board-blue-pill-with-arduino-bootloader.html
Upload over dfu doesn't work, same error messages.

ubis, 2018-02-20:
Hmm, wierd. It could be that it's related with dfu-util, due it's error `Failed to open serial device`.
Unfortunately, I don't have OS X to reproduce this issue.

If I recal correctly, @ivankravets uses OS X and has some generic STM32F103 boards laying around?

ivankravets, 2018-12-11:
We have just updated stm32duino upload tools. Please switch to the latest upstream version https://docs.platformio.org/en/latest/platforms/ststm32.html#upstream and run `pio update`.

Does it work now?

closed 2019-02-21 by ivankravets