platformio/platform-ststm32 #38: Wrong linker script used for generic_stm32f103c ( with fix proposal )

maichaell, 2017-03-28:
Problem was that platformio builded and uploaded (dfu) the flash file OK but it did not boot. 

Reason seems to be platformio links files with linker script 'jtag.ld'  

Arduino IDE build uses 'bootloader_20.ld' script 

-T"bootloader_20.ld"

So I added the bolded rows into arduino.py and it works again. 

I don't know if this is the right way but obviously the row `board.get("upload.boot_version", 0) == 2:` did not work for generic_stm32f103c  and DFU upload option.

    if "stm32f103r" in board.get("build.mcu", ""):
            env.Replace(LDSCRIPT_PATH="bootloader.ld")
        **elif "stm32f103c" in board.get("build.mcu", ""):
            env.Replace(LDSCRIPT_PATH="bootloader_20.ld")**
        elif board.get("upload.boot_version", 0) == 2:
            env.Replace(LDSCRIPT_PATH="bootloader_20.ld")

This issue has been a blocker for me in using platformio for many months but now I finally am able to use it again !

stevstrong, 2017-04-16:
You can add to the board definition file:
``` 
"upload": {
    "protocol": "dfu",
    "boot_version": 2
  },
```
then it should work without changing arduino.py.

v01d, 2017-06-20:
Hi, I was trying to use the Blue Pill board uploading via USB DFU bootloader and I stumbled upon this issue. I was following a thread on the stm32duino forum, where it was mentioned that the problem with DFU was wrong linker script. I found in that thread a modified board definition file (http://www.stm32duino.com/viewtopic.php?f=41&t=728&start=50#p25832) however, specifying the linker script there does not seem to work (it still selects the jtag_c8.ld). Only by modifying arduino.py it uses the correct linker script.

v01d, 2017-06-20:
Ah, sorry, I know see that the suggested change in the board definition file makes it select the linker script.

In any case, could this be changed for the shipped json file? This why DFU would work out of the box.

stevstrong, 2017-06-20:
You can issue a PR here: https://github.com/platformio/platform-ststm32/
My PR regarding support also for F4 is ongoing.

v01d, 2017-06-20:
Ok, I'll try to make a PR. The change from "serial" to "dfu" is OK? Is that the default which would be applied for the board?

Also, Arduino IDE invokes the reset.py script which allows the board to reset via the CDCACM device (by sending a specific sequence via serial commands) into the bootloader. How would this be achieved (invoking the reset.py script) in platformio?

commit 25d48511 (2017-07-19): Change stm32f103c generic boards bootloader version (#57) // Resolve #53  Resolve #38
closed 2017-07-19 by ivankravets