platformio/platform-ststm32 #42: framework: arduino - using ST-Link probe will not reset the board after upload -

stevstrong, 2017-04-16:
Hello,

I am using an ST-Link probe for uploading on STM32F103 and F407 boards.
The problem is that the boards will not be reset after upload.

The root cause seems to be the used flash utility `"st-flash.exe"` from` ".platformio\packages\tool-stlink"` folder.

The Arduino IDE uses `"ST-LINK_CLI.exe"` from the `.platformio\packages\tool-stm32duino\stlink` folder, and it can successfully reset the boards after uploading.

Is it possible to use the same utility for ST-Link for uploading? Or at least make an option to select an appropriate upload tool (with parameters) via board definition file?

Thank you in advance.

valeros, 2017-04-18:
Hi @stevstrong !
You could try to restart your boards after uploading with using [extra_script](http://docs.platformio.org/en/latest/projectconf.html#projectconf-extra-script) option:

Create an extra script file (e.g. `reset_after_upload.py`) beside `platformio.ini`  in your project directory:
```python
Import('env')

env.Prepend(
  UPLOADERFLAGS=["--reset"]
)
```
Add `extra_script` option to your `platformio.ini`:
```ini
[env:disco_f407vg]
platform = ststm32
framework = mbed
board = disco_f407vg
extra_script = reset_after_upload.py
```

stevstrong, 2017-04-18:
Well, I tried your suggestion, but it did not work.

[image]

valeros, 2017-04-18:
I see you are using a custom F4 board configuration. Is the reset pin `NRST` from stlink header connected to a target mcu?

stevstrong, 2017-04-18:
Yes, but it is not connected to the ST-Link probe. Only 3 lines (SWDIO, SWCLOCK, GND) are connected to the probe.

The reset signal will be normally sent over SWD protocol. This is handled correctly by "ST_LINK_CLI.exe" used by Arduino.

valeros, 2017-04-18:
According to Arduino `stlink_upload.bat` next extra script should work:
```python
from os.path import join

Import('env')

env.Replace(
    UPLOADER=join(env.PioPlatform().get_package_dir("tool-stm32duino"), "stlink", "ST-LINK_CLI.exe"),
    UPLOADERFLAGS=[
        "-c",
        "SWD",
        "-P",
        "$SOURCES",
        "0x08000000",
        "-Rst",
        "-Run"
    ]
)
```

stevstrong, 2017-04-18:
Thank you, now it works!

```
Auto-detected: COM6
Uploading .pioenvs\genericSTM32F407VE\firmware.bin
STM32 ST-LINK CLI v2.1.0
STM32 ST-LINK Command Line Interface
 
ST-LINK SN : 57FF6F065077545136281587
ST-LINK Firmware version : V2J27S6

Connected via SWD.
SWD Frequency = 1800K.
Connection mode : Normal.
Device ID:0x413
Device flash Size : 512 Kbytes
Device family :STM32F40xx/F41xx
 
Loading file...
Flash Programming:
File : .pioenvs\genericSTM32F407VE\firmware.bin
Address : 0x08000000
Flash memory programming...
 0%�������������������������������������������������� 100%
Flash memory programmed in 0s and 844ms.
Programming Complete.
 
MCU Reset.
 
Application started.
 
 [SUCCESS] Took 4.16 seconds
```

Would it be possible to integrate it to `"arduino.py"` ?

stevstrong, 2017-04-18:
I replaced the old UPLOADER lines in main.py with these ones ans it still works. I will keep it this way in my repo, thanks.

If you think it makes sense to change it in platformio repo as well, then I could prepare a PR.

ivankravets, 2018-01-24:
Does it work with the latest development version?
```
[env:...]
platform = https://github.com/platformio/platform-ststm32.git
...
```

stevstrong, 2018-01-24:
I will test it in the next days and give feedback.

ivankravets, 2018-02-19:
It should work with the latest dev version

closed 2018-02-19 by ivankravets

spcplugins, 2021-10-13:
I have exactly the same issue in 2021. The solution of adding a .py file given above works, but I assume that the fix to Platform IO has been somehow left out, or modified in some way?