platformio/platform-ststm32 #43: Upload with ST-Link 2.1 fail after firmware upgrade of ST-Link

sstaub, 2017-04-20:
The upload with ST-Link 2.1 to STM32 Nucleo boards fails after firmware upgrade of the ST-Link.
ST-Link 2.1 works with Version V2J25M14, but fails with firmware V2J27M15 and later.
I have this problem with Nucleo- STM32F303K8 and  STM32F432KC. I have also another Nucleo STM32F303 with ST-Link firmware V2J25M14 and it works without any problem. The latest firmware of ST-Link is V2J28M18

stevstrong, 2017-04-20:
Check out [this](https://github.com/platformio/platform-ststm32/issues/42#issuecomment-294798742) and [this](https://github.com/platformio/platform-ststm32/issues/42#issuecomment-294887270) post.

sstaub, 2017-04-20:
These boards are mbed

stevstrong, 2017-04-20:
Just use your board definition and Arduino framework adapt the rest (extra script).

sstaub, 2017-04-20:
I'm working with these boards on mbed framework not with Arduino. I don't understand why it works with an older firmware of ST-Link and not with the newer one.

stevstrong, 2017-04-20:
What kind of error message you get? Can you post it here?

sstaub, 2017-04-20:
See at Forum [https://community.platformio.org/t/upload-to-nucleo-stm32-l423kc-fails/1783](url)

stevstrong, 2017-04-20:
Wrong link?

sstaub, 2017-04-20:
That the right link [https://community.platformio.org/t/upload-to-nucleo-stm32-l423kc-fails/1783](url)

sstaub, 2017-04-20:
new link https://community.platformio.org/t/upload-to-nucleo-stm32-l423kc-fails/1783

ivankravets, 2017-04-29:
Could you try the new ST-Link v1.3.1?

1. `pio platform uninstall ststm32`
2. `pio platform install https://github.com/platformio/platform-ststm32.git`

If you already have GIT version fo ST STM32 development platform, then just update it via `pio update`.

sstaub, 2017-04-29:
Thank you, it works. I tested with Nucleo STM32L423KC :) :) :)

ivankravets, 2017-04-29:
Thanks a lot for the report!

closed 2017-04-29 by ivankravets