platformio/platform-ststm32 #45: Extending libopenCM3 support for other processor

krish2487, 2017-04-29:
Hello,
I would like to thank the team for the wonderful work and prompt response shown to me and many others like me on the forum.

I have yet another request that I believe will make platformio much more potent and feature rich.
The present package for libopenCM3 even in platformio contains the processors and families that is mostly uptodate. 

However, the platformio IDE itself has limited support for the number of processors and boards.
Right now, I am having to deviate away from platformio and use the makefiles method to work with libopencm3. I would like to do it within platformio IDE itself (The peripheral support and features make it too hard to move away from platformio :-D ).

I would like to request the team to see if they can make the build script a little more generic so as to include the processors and families that are supported by libopenCM3. 
I think many more users will be eager to see the extended support for the processors.

@valeros  @ivankravets 
Thank you and have a nice day!! :-)

krish2487, 2017-04-30:
@valeros 
Hello,
@ivankravets  asked me to reach out to you in case I had some specific query. I understand that this development must be taking a significant time and effort on your part. I would like to step forward in anyway I can. 

I am not too good with the scons build system. If you can educate me a little I ll do my best to pitch in with what I can do to contribute.

As I understand it, the libopenCM3 version used by platformio has the the entire list of supported processors. However the entire list is not reflected in the platformio IDE.

If you can spare some time and educate me how the present build script works, section by section I ll try to modify and push the changes as far as I can.

Thank you and have a nice day!! :-)

krish2487, 2017-08-29:
Hello,

any ETA on libopencm3 support. :-)
As I said, if anyone is willing to handhold me to understand how the present build script works, I can attempt to fix it myself.

:-)

Thank you again!

krish2487, 2017-09-01:
Okay, a bit of tinkering and I have progressed thus far. I have used the extra_scripts option to replace the environment variables to suit another processor and tried to run the build command. It still is giving me an error. The error being it is still searching for a F1 processor family to compile whereas the code is written for an F0 processor and the environment variables are modified for a F0 processor. I am assuming I am not seeing a variables or a switch somewhere that is causing the system to build for a F1 processor instead of a F0. 

@valeros  @ivankravets  I am sorry for disturbing you guys like this. Can you help me with this please. Some other section where I am not configuring the processors and or defines properly that is breaking the build. I think the build system can be expanded to other processors for libopencm3. 
The board I am using is a stm32F030R8T6 discovery board. I 

This is contents of my customscript.py
```
Import ('env')
print 80*'='
print 80*'='
print 80*'='
env.Replace(BOARD_MCU=u'stm32f030r8t6')
env.Replace(CPPDEFINES=[('F_CPU','$BOARD_F_CPU'),(u'STM32F0',),('PLATFORMIO',30500),('STM32F03XZ6')])
env.Replace(ASFLAGS=['-x','assembler-with-cpp', '-g','-Os','-ffunction-sections', '-fdata-sections', '-Wall', '-mthumb', '-nostdlib', u'-mcpu=cortex-m0'])
env.Replace(CCFLAGS=['-g','-Os','-ffunction-sections', '-fdata-sections', '-Wall', '-mthumb', '-nostdlib', u'-mcpu=cortex-m0'])
env.Replace(LDSCRIPT_PATH=u'/Users/srikrishnachaitanyanarumanchi/.platformio/packages/framework-libopencm3/lib/stm32/f0/stm32f03xz6.ld')
env.Replace(LINKFLAGS=['-Os', '-Wl,--gc-sections,--relax', '-mthumb', '-nostartfiles', '-nostdlib', u'-mpcu=cortex-m0', '-Wl,-T"$LDSCRIPT_PATH"'])
print 'BOARD_MCU:',env.Dump("BOARD_MCU")
print 'CPPDEFINES:',env.Dump("CPPDEFINES")
print 'ASFLAGS:',env.Dump("ASFLAGS")
print 'CCFLAGS:',env.Dump("CCFLAGS")
print 'LDSCRIPT_PATH:',env.Dump("LDSCRIPT_PATH")
print 'LINKFLAGS:',env.Dump("LINKFLAGS")
print 80*'='
print 80*'='
print 80*'='

```

This is the content of my platformio.ini

```
[env:nucleo_f103rb]
platform = ststm32
board = nucleo_f103rb
framework = libopencm3
build_flags = -DSTM32F0
extra_scripts = customscript.py
```

This is my main.cpp

```
#include <libopencm3/stm32/rcc.h>
#include <libopencm3/stm32/gpio.h>

#define RCCLEDPORT (RCC_GPIOC)
#define LEDPORT (GPIOC)
#define LEDPIN (GPIO9)

static void gpio_setup(void)
{
	/* Enable GPIO clock. */
	/* Using API functions: */
	rcc_periph_clock_enable(RCCLEDPORT);
	/* Set pin to 'output push-pull'. */
	/* Using API functions: */
	gpio_mode_setup(LEDPORT, GPIO_MODE_OUTPUT, GPIO_PUPD_NONE, LEDPIN);
}

int main(void)
{
	int i;
	gpio_setup();
	/* Blink the LED on the board. */
	while (1) {
		/* Using API function gpio_toggle(): */
		gpio_toggle(LEDPORT, LEDPIN);	/* LED on/off */
		for (i = 0; i < 1000000; i++) {	/* Wait a bit. */
			__asm__("nop");
		}
	}

	return 0;
}

```

And this is the error message screenshot when I try to run pin build.

[image]
[image]
[image]

krish2487, 2017-09-02:
Well then, finally got it compile the flash for STM32F030R8 processor.
The changes I had to make were so.
1. The disco_f030r8.json file had to be modified to this

```
  "build": {
    "core": "stm32",
    "cpu": "cortex-m0",
    "extra_flags": "-DSTM32F030x8",
    "f_cpu": "48000000L",
    "mcu": "stm32f030r8t6",
    "ldscript": "stm32f03xz6.ld",
    "variant": "stm32f0"
  },
  "frameworks": [
    "mbed",
    "stm32cube",
    "libopencm3"
  ],
```
Basically, manually defining the linker script and defining the variant and MCU.

@valeros  @ivankravets  
I might have gone overkill for this, but is it just enough if the board json was modified to extend the libopencm3 support??

krish2487, 2017-09-02:
And I can confirm that making the same changes to the nucleo_f303re board.json also works! :-)
Maybe we can compile the boards that support the libopencm3 framework and make the necessary changes.

The immediate issue I see is if another user is using another framework for coding on the same processor, it is most likely to fail owing to the hardcoded linkerscripts and such. Maybe there is a way to customize the json file definitions as per the framework??

valeros, 2020-08-03:
Resolved in https://github.com/platformio/platform-ststm32/commit/0bf8d98f1944d76e4b8159962c04969d610c30ab

closed 2020-08-03 by valeros