platformio/platform-ststm32 #47: STM32F103 executable size optimization

Doowybbob, 2017-06-04:
I've been working with a Nucleo-F103 development board and am trying to move over to PlatformIO for development and testing. 

However, I've noticed that PlatformIO generates an executable about twice the size of my typical development environment which is gcc4mbed (https://github.com/adamgreen/gcc4mbed). PlatformIO produces an executable that is 92KB and gcc4mbed produces an exectable that is 51KB.

Both are using arm-none-eabi-g++ so I'm guessing it's a problem with compiler flags or libraries that are being included? Digging into the flags a bit, I haven't found any obvious causes for the size increase. PlatformIO uses the -Os flag, and gcc4mbed uses -O2 but they should produce similar code sizes.

I've attached a log file from compiling in both environments if that helps at all.

[piolog.txt](https://github.com/platformio/platform-ststm32/files/1050467/piolog.txt)
[makelog.txt](https://github.com/platformio/platform-ststm32/files/1050470/makelog.txt)

valeros, 2017-06-05:
Hi @Doowybbob!
To reduce the binary size you can use newlib-nano specs. The easiest way is to add an extra script to the root folder of project (e.g. `add_nanolib.py`):
```python
Import("env")
env.Append(LINKFLAGS=["--specs=nano.specs"])
```
And update your `platformio.ini`:
```ini
[env:nucleo_f103rb]
platform = ststm32
framework = mbed
board = nucleo_f103rb
extra_script = add_nanolib.py
```

Doowybbob, 2017-06-05:
Awesome, that brought the size down quite a bit!
And I learned about extra scripts - looks powerful.

closed 2018-02-19 by ivankravets

bakwc, 2021-06-01:
Tried and got error:
`attempt to rename spec 'link' to already defined spec 'nano_link'`

13rac1, 2022-02-20:
@bakwc You've probably solved, but in case someone else finds this: [[src](https://community.nxp.com/t5/Kinetis-Design-Studio/elf-error/td-p/428746)]
> This is because you have specified nano.specs twice: "--specs=nano.specs -specs=nano.specs" in you command line.

Build log:
```
$ platformio run -v
<SNIP>
Building in release mode
arm-none-eabi-gcc -o .pio/build/STM32F0XXXXXX/firmware.elf -T /home/user/.platformio/packages/tool-ldscripts-ststm32/stm32f0/STM32F0XXXXXX_FLASH.ld -Os -Wl,--gc-sections,--relax -mthumb -mcpu=cortex-m0 --specs=nano.specs --specs=nosys.specs --specs=nano.specs .pio/build/STM32F0XXXXXX/FrameworkHALDriver/Src/stm32f0xx_hal.o <SNIP>
```
Ah yeah, it's set twice.