platformio/platform-ststm32 #48: REQUEST: Update of framework-arduinoststm32

noisymime, 2017-06-05:
Recently there has been a lot of pull requests brought into master on the base framework for stm32duino: https://github.com/rogerclarkmelbourne/Arduino_STM32

These have added significant functionality and additional compatibility with the arduino API. 

It looks like the last pull from this for package framework-arduinoststm32 2.0.0 was in Oct 2016.

Would it be possible to get a revised package version pulled from master on the above repo?

ivankravets, 2017-07-23:
@ubis Could we update to the latest source code of stm32duino? Should we change build script?

ubis, 2017-07-23:
I believe so, yes. I've looked at newer `boards.txt` and `platform.txt` to see changes. There wasn't any major changes at all.
So, it seems like no changes to stm32duino should be done.

ivankravets, 2017-07-23:
@ubis Thank you very much! I've just updated a package. 

@noisymime please `pio update`.

closed 2017-07-23 by ivankravets

noisymime, 2017-07-23:
Tested and working. Huge thanks @ivankravets !