platformio/platform-ststm32 #51: mbed, SPL, arduino support for STM32F411E-DISCO

jlunder, 2017-06-17:
I have a couple STM32F411E-DISCO boards, which are based on the STM32F411VET6. This board doesn't seem to be in the database!

I'd be happy to help out with a pull request if someone can point to guidelines and give me an idea where to start.

valeros, 2017-06-19:
Hi @jlunder!
As for mbed, there is no support for STM32F411E-DISCO for some reason:
https://github.com/ARMmbed/mbed-os/issues/2390
https://developer.mbed.org/platforms/ST-Discovery-F411E/
Same for Arduino, we have a limited board list with Arduino feature for STM32.
As for SPL, we decided to move to newer and more modern STM32Cube HAL framework.

jlunder, 2017-06-21:
So basically raise the issue with upstream for mbed/STM32duino? I can do that. I've been using STM32Cube HAL and not SPL, so I would be thrilled to see that library supported with this board.

closed 2017-06-21 by jlunder