platformio/platform-ststm32 #53: upload via STM32duino bootloader

universam1, 2017-07-03:
Hi,
I tried all kinds of hints from other issues, but cannot get upload via `STM32duino bootloader` working in platformio. **Inside Arduino it works flawless!**

I got installed:
```
ststm32 @ 3.0.0
framework-arduinoststm32 @ 2.0.0
tool-stlink @ 1.10301.0
tool-stm32duino @ 1.0.0
toolchain-gccarmnoneeabi @ 1.40804.0
```
platformio.ini:
```
[env:STM]
platform = ststm32
framework = arduino
board = bluepill_f103c8
```

result:
`s2t-flash 1.3.1017-07-03T13:39:09 WARN src\usb.c: Couldn't find any ST-Link/V2 devices`
Well, makes some sense although I would expect that it uses the STM32duino bootloader...

So if I change then `upload_protocol = serial` the result is:
```Looking for upload disk...
Auto-detected: COM13
Uploading .pioenvs\STM\firmware.bin
stm32flash 0.4Failed to init device.

http://stm32flash.googlecode.com/

Using Parser : Raw BINARY
Interface serial_w32: 230400 8E1
```
This doesn't seam right to me since the upload process from Arduino is different and uses a `java -jar maple_loader.jar %1 %2 %3 %4 %5 %6 %7 %8 %9` uploader:

```
C:\arduino-1.6.11\hardware\Arduino_STM32/tools/win/maple_upload.bat COM13 2 1EAF:0003 C:\Users\sl\AppData\Local\Temp\build0fed3136074a62d7fa3761becf2703b1.tmp/StressSerialUSB.ino.bin 
maple_loader v0.1
Resetting to bootloader via DTR pulse
Searching for DFU device [1EAF:0003]...
Found it!

Opening USB Device 0x1eaf:0x0003...
Found Runtime: [0x1eaf:0x0003] devnum=1, cfg=0, intf=0, alt=2, name="STM32duino bootloader v1.0  Upload to Flash 0x8002000"
Setting Configuration 1...
Claiming USB DFU Interface...
Setting Alternate Setting ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
Transfer Size = 0x0400
bytes_per_hash=417
Starting download: [##################################################] finished!
error resetting after download: usb_reset: could not reset device, win error: The system cannot find the file specified.
state(8) = dfuMANIFEST-WAIT-RESET, status(0) = No error condition is present

Done!

Resetting USB to switch back to runtime mode
```

Any idea how to do it?

Thanks!

universam1, 2017-07-05:
Okay it is possible to upload if I set the board to `maple mini` and suddenly the STM32duino bootloader is chosen.
But anyway, this not correct since the chip is not the same.
I though cant select DFU as upload method on the bluepill_f103c8 board.

ubis, 2017-07-05:
There are multiple upload protocols(dfu, jtag, serial, ...).
Use `upload_protocol = dfu` in your platformio.ini file in order to use dfu flashing option(STM32Duino bootloader).

universam1, 2017-07-05:
Thank you for your reply, I tried again with `upload_protocol = dfu` and it fails like this:

```
KeyError: "Invalid board option 'build.hwids'":
File "C:\python27\lib\site-packages\platformio\builder\main.py", line 155:
env.SConscript("$BUILD_SCRIPT")
File "C:\Users\Sam\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
return _SConscript(self.fs, *files, **subst_kw)
File "C:\Users\Sam\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
exec _file_ in call_stack[-1].globals
File "C:\Users\Sam\.platformio\platforms\ststm32\builder\main.py", line 147:
_usbids = env.BoardConfig().get("build.hwids")
File "c:\python27\lib\site-packages\platformio\managers\platform.py", line 617:
raise KeyError("Invalid board option '%s'" % path)
```
Could you please try, I tried on two different computer same result.

ubis, 2017-07-06:
Hmm, we forgot to add hwids to bluepill board. I will make a PR to fix this.
In the meantime, could you use `board = genericSTM32F103C8` in your platformio.ini?

universam1, 2017-07-06:
Great!
Actually, if I use `board = genericSTM32F103C8` the upload works, but the board crashes. Tested all kinds of combination, only declaring it as `maple` produces valid code.

Tested also with pure Arduino Studio, and there choosing Generic STM32F103C works correctly. So there must be another problem.

ubis, 2017-07-06:
Hmm weird, maple uses different mcu(F103RB) instead of bluepill's F103C8.
Could you post your Arduino IDE build & upload logs somewhere, like pastebin?

universam1, 2017-07-19:
Hi, sorry for delay. I've tested all way down with the latest 3.4.1a6, I see above PR is in, nice!

So when compiling the `blink` example with Arduino, selecting 'Generic STM32F103C' as board with 64k flash, everything is fine.
https://pastebin.com/aDLpf8Xq

Now, the same source code with PIO 
```
[env:bluepill]
platform = ststm32
board = bluepill_f103c8
framework = arduino
upload_protocol = dfu
``` 
compiles, upload does now work (:heart:) but totally crashed the device:
https://pastebin.com/vjeB96S0
The same happens if I choose `board = genericSTM32F103C8`
That means, I have to 'resurrect' the mcu via arduino.

Now, as being said above if I pretend this being a `board = maple_mini_b20` it actually does compile, upload and work nicely!
So the compiler parameter must be broken.

Thanks

ubis, 2017-07-19:
Seems like wrong linker script is used. Could you edit manually `bluepill_f103c8.json` file and type in `upload` section: `"boot_version": 2`, so it would look like this:

```
  "upload": {
    "boot_version": 2,
    "maximum_ram_size": 20480,
    "maximum_size": 65536,
    "protocol": "stlink"
  },
```

And report back? Thanks.

universam1, 2017-07-19:
Awesome, I can confirm that it does work with your mentioned changes of the linker!
I can now compile, upload and it runs correctly!
👍 

Great work, hope it gets into master soon!

closed 2017-07-19 by ivankravets

VSG24, 2020-08-30:
This issue still exists. It shouldn't be closed.
I'm using the latest version of everything and also changed `bluepill_f103c8.json` file accordingly and added `boot_version` however that changed nothing. Still uploads fine but corrupts the bootloader.

**The only solution that works is defining `board = maple_mini_b20` and `upload_protocol = dfu`  in `platformio.ini` file.**

_Why is this still not fixed after +3 years?_

rob-deutsch, 2020-12-06:
I can confirm that, as per @VSG24, I am having exactly the same problem.

The code seems to run with `board = bluepill_f103c8` but it totally ruins the USB serial port.

rob-deutsch, 2020-12-08:
A-ha! Progress! There are some important `build_flags`.

I have been trying to replicate the experience I get with the Arduinio IDE and the following URLs...

https://github.com/rogerclarkmelbourne/Arduino_STM32
http://dan.drown.org/stm32duino/package_stm32duino_index.json

... and I have finally got there with this:

````
[env:bluepill_f103c8]
platform = ststm32
board = bluepill_f103c8
framework = arduino
board_build.core = maple
build_flags = -DSERIAL_USB -DGENERIC_BOOTLOADER
upload_protocol = dfu
````

JakubRakus, 2021-01-11:
Confirm, issue still exists, can't upload to blue/blackpill STM32F103 based board (with 1k5 pullups on USB D+ line) using DFU and bootloader from https://github.com/rogerclarkmelbourne/STM32duino-bootloader

No matter which core (maple/arduino) is used. I've tried to change stable versions of `platform` and `board_build.core` to the development ones from github - nothing changes.

Bootloader presents itself as DFU device (pid=0003) only a few seconds after reset, then switch to serial port (pid=0004), so `maple_upload` script should first reset STM device using `upload-reset` tool, wait some milliseconds and then load new app using `dfu-util`. The problem lies in the `upload-reset` tool - it crashes with various errors trying to `setRTS()` or `setDTR()`.

MURPHYENGINEERING, 2022-01-01:
Currently having the same problem with generic STM32F103C6T6 board. The arduino IDE works fine, but platformio never finds a DFU device. I tried all the examples given in this thread.

commit 25d48511 (2017-07-19): Change stm32f103c generic boards bootloader version (#57) // Resolve #53  Resolve #38