platformio/platform-ststm32 #55: mbed-filesystem compile error

fidomax, 2017-07-10:
------------------------------------------------------------------

### Configuration

**Operating system**:Ubuntu 17.04 x64

**PlatformIO Version** (`platformio --version`): PlatformIO, version 3.4.1a5

### Description of problem

Failed to compile mbed-filesystem

#### Steps to Reproduce

1. Create any mbed project from example
2. Add #include "FATFileSystem.h" at main.cpp
3. Try to compile

### Actual Results

[Mon Jul 10 12:47:23 2017] Processing nucleo_f411re (upload_protocol: stlink; platform: ststm32; board: nucl
eo_f411re; framework: mbed)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Collected 8 compatible libraries
Looking for dependencies...
￼
Library Dependency Graph
|-- <mbed-filesystem>
|-- <mbed-rtos>
Compiling .pioenvs/nucleo_f411re/src/main.o
Compiling .pioenvs/nucleo_f411re/lib/filesystem/Dir.o
Compiling .pioenvs/nucleo_f411re/lib/filesystem/File.o
In file included from /home/maximus/.platformio/packages/framework-mbed/features/filesystem/Dir.cpp:17:0:
/home/maximus/.platformio/packages/framework-mbed/features/filesystem/Dir.h:20:35: fatal error: filesystem/F
ileSystem.h: No such file or directory
#include "filesystem/FileSystem.h"
^
compilation terminated.
Linter
Severity 	Provider 	Description	Line 
Git
GitHub (preview)
PIO Buildsrc/main.cpp0208:1(1, 26)
LFUTF-8C++0 files

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

ivankravets, 2017-07-23:
Do you have problem with the latest updates?

valeros, 2017-07-24:
Hi @fidomax !
Could you please retest your example with latest `ststm32` platform?

fidomax, 2017-07-24:
Still not working...

Compiling .pioenvs/nucleo_f411re/lib/filesystem/Dir.o
Compiling .pioenvs/nucleo_f411re/lib/filesystem/File.o
Compiling .pioenvs/nucleo_f411re/lib/filesystem/FileSystem.o
Compiling .pioenvs/nucleo_f411re/lib/filesystem/bd/ChainingBlockDevice.o
Compiling .pioenvs/nucleo_f411re/lib/filesystem/bd/HeapBlockDevice.o
/home/maximus/.platformio/packages/framework-mbed/features/filesystem/FileSystem.cpp: In instantiation of 'class Managed<mbed::File>':
/home/maximus/.platformio/packages/framework-mbed/features/filesystem/FileSystem.cpp:139:19:   required from here
/home/maximus/.platformio/packages/framework-mbed/features/filesystem/FileSystem.cpp:128:7: error: invalid use of incomplete type 'class mbed::File'
class Managed : public F {
^~~~~~~
In file included from /home/maximus/.platformio/packages/framework-mbed/features/filesystem/FileSystem.cpp:18:0:
/home/maximus/.platformio/packages/framework-mbed/features/filesystem/FileSystem.h:39:7: note: forward declaration of 'class mbed::File'
class File;
^~~~
/home/maximus/.platformio/packages/framework-mbed/features/filesystem/FileSystem.cpp: In member function 'virtual int mbed::FileSystem::open(mbed::FileHandle**, const char*, int)':
/home/maximus/.platformio/packages/framework-mbed/features/filesystem/FileSystem.cpp:139:19: error: cannot convert 'Managed<mbed::File>*' to 'mbed::File*' in initialization
File *f = new Managed<File>;
^~~~~~~~~~~~~
/home/maximus/.platformio/packages/framework-mbed/features/filesystem/FileSystem.cpp:140:16: error: invalid use of incomplete type 'class mbed::File'
int err = f->open(this, path, flags);
^~
In file included from /home/maximus/.platformio/packages/framework-mbed/features/filesystem/FileSystem.cpp:18:0:
/home/maximus/.platformio/packages/framework-mbed/features/filesystem/FileSystem.h:39:7: note: forward declaration of 'class mbed::File'
class File;
^~~~
/home/maximus/.platformio/packages/framework-mbed/features/filesystem/FileSystem.cpp:142:16: warning: possible problem detected in invocation of delete operator: [-Wdelete-incomplete]
delete f;
^
/home/maximus/.platformio/packages/framework-mbed/features/filesystem/FileSystem.cpp:139:11: warning: 'f' has incomplete type

valeros, 2017-07-24:
Could you please try to add this line to your `platformio.ini`:
```
build_flags = -D PIO_FRAMEWORK_MBED_FILESYSTEM_PRESENT
```

fidomax, 2017-07-24:
Now it works! 

Generating LD script .pioenvs/nucleo_f411re/STM32F411XE.ld.link_script.ld
Linking .pioenvs/nucleo_f411re/firmware.elf
Building .pioenvs/nucleo_f411re/firmware.bin
Calculating size .pioenvs/nucleo_f411re/firmware.elf
text       data     bss     dec     hex filename
9836       2140    1332   13308    33fc .pioenvs/nucleo_f411re/firmware.elf

valeros, 2017-07-24:
Don't forget to check other configuration options for mbed framework [here](http://docs.platformio.org/en/latest/frameworks/mbed.html#configuration).

closed 2017-07-24 by valeros