platformio/platform-ststm32 #58: stlink with Nucleo-f446re

NathanielAJohnson, 2017-07-23:
I am receiving the following error when uploading.

```
st-flash: error while loading shared libraries: libstlink-shared.so.1: cannot open shared object file: No such 
file or directory
*** [upload] Error 127
```

This is on Atom version running on Ubuntu 17.04 .  I have found this link to the [same error in the stlink repository](https://github.com/texane/stlink/issues/478).

I found where platformio stores these tools and ran them.  In a system terminal they reported the same error but in the platformio spawned terminal they worked.  The suggestions offered there did not work for obvious reasons.  Leads me to believe the pio environment needs to cloned into that build process.

tobykurien, 2017-09-20:
Problem appears to be that platformio is executing `tool-stlink/bin/st-flash` instead of `tool-stlink/st-flash`

ivankravets, 2017-09-20:
Fixed. Please run `pio update` command.

closed 2017-09-20 by ivankravets

tingwu88, 2018-01-16:
but when i run the command, it shows me that no command 'pio' found ??

Slavjan, 2018-04-28:
this may be fixed via `sudo ldconfig`, and nothing else.

Slavjan, 2018-04-28:
this may be fixed via `sudo ldconfig`, and nothing else. 
What is a `pio` ? I can`t found that...

gustawdaniel, 2018-11-05:
@ivankravets what is `pio`, where should i run this command?

ivankravets, 2018-11-06:
@gustawdaniel do you have this issue? What is your OS?

gustawdaniel, 2018-11-08:
Ubuntu 18.04

I fixed by `sudo ldconfig` executed in directory when I was compiling library.

Basilisvirus, 2020-03-13:
Hello,
ld-config does not solve my problem. (i do not use pio either)

My error is: 
`st-info: error while loading shared libraries: libstlink.so.1: cannot open shared object file: No such file or directory`

in issue #478 i also just posted the same error i get and what i tried to do to solve it.