platformio/platform-ststm32 #59: SDBlockDevice and FATFileSystem not working

mcprakash, 2017-07-25:
Hi 
 I am trying  to do a simple SPI read/write to an SD card using the following libraries

> SDBlockDevice.h
> FATFileSystem.h

 I am having trouble with the `-D PIO_FRAMEWORK_MBED_FILESYSTEM_PRESENT` flag. 

 The exact output message is attached in screenshot.
[image]

 However, if I go to the `mbed-os` github repository for `v5.50`, I am able to see the `FATFileSystem` library.

 Could you help me with this ? Thanks

Chandra

mcprakash, 2017-08-01:
@ivankravets @valeros I tried a simple hello world program with the flag -D PIO_FRAMEWORK_MBED_FILESYSTEM_PRESENT and I get compile time error. PFA the attahced snapshot for further reference. It will be great if you can help me out. 

[image]

DL6AKU, 2018-05-03:
I am having the same problem here.

```
.platformio/packages/framework-mbed/mbed.h:50:10: fatal error: filesystem/mbed_filesystem.h: No such file or directory
#include "filesystem/mbed_filesystem.h"
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/nucleo_l476rg/FrameworkMbedCore/platform/FileSystemHandle.o] Error 1
```

mcprakash, 2018-05-03:
Hi DL6AKU, 
 I switched to mbed command line tool and it works without any issues. I think there is a problem with ST-NUCLEO devices and SD FILE System libraries when used together with platformio in mbed environment.
 Hope this helps. 
Best
Chandra

closed 2018-05-03 by mcprakash