platformio/platform-ststm32 #60: Flashing problem

kwetter, 2017-07-28:


kwetter, 2017-07-28:
Seems i got it working for the Maple Mini Clone (STM32F103CBT6) with following init file:
[env:maple_mini_origin]
platform = ststm32
board = maple_mini_origin
framework = arduino
upload_protocol = dfu

problem persists though with the BluePill (STM32F103C8T6) and this init file:
[env:genericSTM32F103C8]
platform = ststm32
board = genericSTM32F103C8
framework = arduino
upload_protocol = dfu

any hints?

kwetter, 2017-07-28:
Got it working, so for what it's worth for others:

The Blue Pill Board (STM32F103C8T6) works for upload via USB with dfu-util installed on OSX with following platformio.ini:

platform = ststm32
board = bluepill_f103c8
framework = arduino
upload_port = anything
upload_protocol = dfu

i had to define upload_port (doesn't matter what...). I still have to reset the board manually right before uploading.

maybe there's a better solution, but I think I'm satisfied for the moment :)

ubis, 2017-08-01:
To able to flash board within dfu, you need to have stm32bootloader installed. Choose the correct one from [here](https://github.com/rogerclarkmelbourne/STM32duino-bootloader/tree/master/STM32F1/binaries). Usually PC13 is port to which LED is connected, so `generic_boot20_pc13.bin` should work in this case.

To flash it, you need usb-ttl converter(ft232rl for example) and set both BOOT's to BOOT1: 0; BOOT0: 1. More info [here](http://www.stm32duino.com/viewtopic.php?t=413).

However, most of these boards contain wrong usb pull-up resistor on D+. More info about it [here](http://wiki.stm32duino.com/index.php?title=Blue_Pill). I had some issues that sometimes it couldn't reset after flash, so i've changed pull-up resistor to 1k5. 

After flashing, set BOOT0 back to 0 and use usb from now on. When creating new project, you can use BluePill board, and set `upload_protocol = dfu`

I'm not sure if you really need to set `upload_port` on unix based systems, but in case it won't automatically find it - you might need it to add it. Thats pretty much it.

kwetter, 2017-08-08:
thanks ubis for the tip about the wrong pull-up resistor. I might fix it with the known work-around.

closed 2017-08-08 by kwetter