platformio/platform-ststm32 #62: Project Initialisation fails when username contains a space

nigelb, 2017-08-10:
Project initialisation for a username with space:

	C:\Users\Billy Bob\Documents\projects\ucontroller\Feral>"c:\Users\Billy Bob\.python\pio\Scripts\pio.exe" init --board mts_mdot_f405rg --ide clion
	****************************************************************************************************************************************************************
	If you like PlatformIO, please:
	- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
	- star it on GitHub > https://github.com/platformio/platformio
	- try PlatformIO IDE for IoT development > http://platformio.org/platformio-ide
	- support us with PlatformIO Plus > https://pioplus.com
	****************************************************************************************************************************************************************

	The current working directory C:\Users\Billy Bob\Documents\projects\ucontroller\Feral will be used for project.
	You can specify another project directory via
	`platformio init -d %PATH_TO_THE_PROJECT_DIR%` command.

	The next files/directories have been created in C:\Users\Billy Bob\Documents\projects\ucontroller\Feral
	platformio.ini - Project Configuration File
	src - Put your source files here
	lib - Put here project specific (private) libraries
	PlatformManager: Installing ststm32
	Downloading  [####################################]  100%
	Unpacking  [####################################]  100%
	ststm32 @ 3.3.0 has been successfully installed!
	PackageManager: Installing toolchain-gccarmnoneeabi @ >=1.40803.0,<1.40805.0
	Downloading  [####################################]  100%
	Unpacking  [####################################]  100%
	toolchain-gccarmnoneeabi @ 1.40804.0 has been successfully installed!
	The platform 'ststm32' has been successfully installed!
	The rest of packages will be installed automatically depending on your build environment.
	TypeError: Tried to lookup File 'C:\Users\Billy Bob\.platformio\platforms\ststm32\builder\frameworks\mbed\Bob\.platformio\packages\framework-mbed\rtos' as a Dir.:
	File "C:\Users\billyb~1\python~1\pio\lib\site-packages\platformio\builder\main.py", line 153:
	env.SConscript("$BUILD_SCRIPT")
	File "C:\Users\Billy Bob\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
	return _SConscript(self.fs, *files, **subst_kw)
	File "C:\Users\Billy Bob\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
	exec _file_ in call_stack[-1].globals
	File "C:\Users\Billy Bob\.platformio\platforms\ststm32\builder\main.py", line 165:
	target_elf = env.BuildProgram()
	File "C:\Users\Billy Bob\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
	return self.method(*nargs, **kwargs)
	File "C:\Users\billyb~1\python~1\pio\lib\site-packages\platformio\builder\tools\platformio.py", line 60:
	env.BuildFrameworks(env.get("PIOFRAMEWORK"))
	File "C:\Users\Billy Bob\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
	return self.method(*nargs, **kwargs)
	File "C:\Users\billyb~1\python~1\pio\lib\site-packages\platformio\builder\tools\platformio.py", line 261:
	SConscript(env.GetFrameworkScript(f))
	File "C:\Users\Billy Bob\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 604:
	return method(*args, **kw)
	File "C:\Users\Billy Bob\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
	return _SConscript(self.fs, *files, **subst_kw)
	File "C:\Users\Billy Bob\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
	exec _file_ in call_stack[-1].globals
	File "C:\Users\Billy Bob\.platformio\platforms\ststm32\builder\frameworks\mbed\mbed.py", line 172:
	get_dynamic_manifest(lib, lib_config, extra_includes))
	File "c:\users\billyb~1\python~1\pio\lib\site-packages\platformio\builder\tools\piolib.py", line 108:
	self.process_extra_options()
	File "c:\users\billyb~1\python~1\pio\lib\site-packages\platformio\builder\tools\piolib.py", line 235:
	self.env.ProcessFlags(self.build_flags)
	File "C:\Users\Billy Bob\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
	return self.method(*nargs, **kwargs)
	File "C:\Users\billyb~1\python~1\pio\lib\site-packages\platformio\builder\tools\platformio.py", line 115:
	parsed_flags = env.ParseFlags(str(flags))
	File "C:\Users\Billy Bob\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 800:
	do_parse(arg)
	File "C:\Users\Billy Bob\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 730:
	dict['LIBS'].append(self.fs.File(arg))
	File "C:\Users\Billy Bob\.platformio\packages\tool-scons\script\..\engine\SCons\Node\FS.py", line 1404:
	return self._lookup(name, directory, File, create)
	File "C:\Users\Billy Bob\.platformio\packages\tool-scons\script\..\engine\SCons\Node\FS.py", line 1383:
	return root._lookup_abs(p, fsclass, create)
	File "C:\Users\Billy Bob\.platformio\packages\tool-scons\script\..\engine\SCons\Node\FS.py", line 2390:
	dir_node = self._lookup_abs(dir_name, Dir)
	File "C:\Users\Billy Bob\.platformio\packages\tool-scons\script\..\engine\SCons\Node\FS.py", line 2403:
	result.must_be_same(klass)
	File "C:\Users\Billy Bob\.platformio\packages\tool-scons\script\..\engine\SCons\Node\FS.py", line 652:
	(self.__class__.__name__, self.get_internal_path(), klass.__name__))
	================================================================= [ERROR] Took 166.80 seconds =================================================================

Project initialisation for a username without a space:

	C:\Users\Bob\Documents\projects\ucontroller\Feral>c:\Users\Bob\.python\pio\Scripts\pio.exe init --board mts_mdot_f405rg --ide clion

	The current working directory C:\Users\Bob\Documents\projects\ucontroller\Feral will be used for project.
	You can specify another project directory via
	`platformio init -d %PATH_TO_THE_PROJECT_DIR%` command.

	The next files/directories have been created in C:\Users\Bob\Documents\projects\ucontroller\Feral
	platformio.ini - Project Configuration File
	src - Put your source files here
	lib - Put here project specific (private) libraries
	PlatformManager: Installing ststm32
	Downloading  [####################################]  100%
	Unpacking  [####################################]  100%
	ststm32 @ 3.3.0 has been successfully installed!
	PackageManager: Installing toolchain-gccarmnoneeabi @ >=1.40803.0,<1.40805.0
	Downloading  [####################################]  100%
	Unpacking  [####################################]  100%
	toolchain-gccarmnoneeabi @ 1.40804.0 has been successfully installed!
	The platform 'ststm32' has been successfully installed!
	The rest of packages will be installed automatically depending on your build environment.

	Project has been successfully initialized!
	Useful commands:
	`platformio run` - process/build project from the current directory
	`platformio run --target upload` or `platformio run -t upload` - upload firmware to embedded board
	`platformio run --target clean` - clean project (remove compiled files)
	`platformio run --help` - additional information

valeros, 2017-08-10:
Hi @nigelb !
This issue has been fixed in dev branch, could you please retest it? Just run next commands in your terminal:
```
pio platform uninstall ststm32
pio platform install https://github.com/platformio/platform-ststm32.git
```

nigelb, 2017-08-11:
Thanks, that works.

closed 2017-08-22 by ivankravets