platformio/platform-ststm32 #64: Adding support for the STM32F030F4

hasxz, 2017-09-03:
While platformio has support for the STM32F030R8, it doesn't have support for the STM32F030F4, even though they are very similar -- only a small difference in periphrials and flash size. Is there some sort of guide for adding support for the board?

valeros, 2017-09-05:
Hi, @hasxz! Is there a development board with this MCU? What framework are you going to use?
You can take the configuration for `STM32F030R8` as an example and provide a PR with changes required for `STM32F030F4`. Here is a [doc page](http://docs.platformio.org/en/latest/platforms/creating_board.html) with info about creating custom boards.

ivankravets, 2019-02-21:
Resolved in https://github.com/platformio/platform-ststm32/blob/develop/boards/demo_f030f4.json

closed 2019-02-21 by ivankravets