platformio/platform-ststm32 #65: mbed-filesystem compile error

eeyurdakul, 2017-09-03:
macOS 10.12.6
PlatformIO 3.5.0a8

compile error when   **#include "SDFileSystem.h"**   is added to main.cpp.
**build_flags = -D PIO_FRAMEWORK_MBED_FILESYSTEM_PRESENT** is added to **platformio.ini** but problem still not solved.

Compile error is as follows:
**/Users/eeyurdakul/.platformio/packages/framework-mbed/mbed.h:51:40: fatal error: filesystem/mbed_filesystem.h: No such file or directory**

valeros, 2017-09-05:
Hi, @eeyurdakul! There is no such header in the framework. You can try [this one](https://developer.mbed.org/handbook/SDFileSystem) but only with `mbed 2`.

eeyurdakul, 2017-09-06:
How can I try it with mbed 2. Do I downgrade the firmware?

valeros, 2017-09-06:
`mbed 2` is enabled by default, therefore you can simply add the SDFileSystem library to your project.

eeyurdakul, 2017-09-06:
Here is the error when I add "#include "SDFileSystem.h" to the project.

/.platformio/packages/framework-mbed/mbed.h:51:40: fatal error: filesystem/mbed_filesystem.h: No such file or directory

eeyurdakul, 2017-09-06:
Sorry, I forgot to add the library. Now it works...Thanks

closed 2017-09-06 by eeyurdakul