platformio/platform-ststm32 #71: Error on compilation error: unknown type name 'bool'

stanleyseow, 2017-10-13:
Tried the latest PlatformIO on Linux (cloud9)  n mac OSX , getting the same error on compilation ...

Using a blank main.c :-
$ cat src/main.c 
#include <Arduino.h>

void setup() {
}

void loop() {
 
}

[Fri Oct 13 12:58:02 2017] Processing genericSTM32F103C8 (platform: ststm32; board: genericSTM32F103C8; framework: arduino)

In file included from .platformio/packages/framework-arduinoststm32/STM32F1/cores/maple/boards.h:38:0,
from .platformio/packages/framework-arduinoststm32/STM32F1/cores/maple/wirish.h:54,
from .platformio/packages/framework-arduinoststm32/STM32F1/cores/maple/Arduino.h:30,
from src/main.c:1:
.platformio/packages/framework-arduinoststm32/STM32F1/cores/maple/wirish_types.h:66:1: error: unknown type name 'bool'
typedef bool boolean;
^
In file included from .platformio/packages/framework-arduinoststm32/STM32F1/cores/maple/wirish.h:54:0,
from .platformio/packages/framework-arduinoststm32/STM32F1/cores/maple/Arduino.h:30,
from src/main.c:1:
.platformio/packages/framework-arduinoststm32/STM32F1/cores/maple/boards.h:111:1: error: unknown type name 'bool'
bool boardUsesPin(uint8 pin);
^
In file included from .platformio/packages/framework-arduinoststm32/STM32F1/cores/maple/wirish.h:58:0,
from .platformio/packages/framework-arduinoststm32/STM32F1/cores/maple/Arduino.h:30,
from src/main.c:1:
.platformio/packages/framework-arduinoststm32/STM32F1/cores/maple/ext_interrupts.h:90:6: error: conflicting types for 'attachInterrupt'
void attachInterrupt(uint8 pin, voidArgumentFuncPtr handler, void *arg,
^

vbenso, 2017-11-05:
Any news on this?

ubis, 2017-11-08:
`unknown type name 'bool'`
`typedef bool boolean;`

There's no such thing as `bool` in C. Rename your `main.c` to `main.cpp`

ivankravets, 2017-11-08:
Thanks, @ubis !

closed 2017-11-08 by ivankravets

tienhung1611, 2020-03-28:
I've got this error too. I'm using PlatformIO on Visual Studio code on Windows 10. My PlatformIO here:
[env:genericSTM32F103VE]
platform = ststm32
board = genericSTM32F103VE
framework = arduino
upload_protocol = stlink
lib_deps = STM32duino FreeRTOS
If I remove "lib_deps = STM32duino FreeRTOS" and remove #include <STM32FreeRTOS.h>, everything fine