platformio/platform-ststm32 #74: Please update tool-stlink

me21, 2017-10-30:
Please update tool-stlink to the latest version. Currently that's 1.4.0 in upstream. The bundled package.json reads version to be 1.10301.0, I'm not sure but it must be lower than upstream.

ivankravets, 2017-12-29:
Could you re-test with?

platformio.ini
```
[env:...]
platform = https://github.com/platformio/platform-ststm32.git
board =...
```

Does it work now?

me21, 2017-12-29:
Ah, I've sorted it out myself already. I'm sorry for not reporting :(
The cause was that I've inserted ST-LINK device into USB 3.0 port. It didn't work with USB 3.0. When I inserted it into USB 2.0 port, everything worked like a charm.

closed 2017-12-29 by me21

me21, 2017-12-29:
Ah, I've sorted it out myself already. I'm sorry for not reporting :(
See https://github.com/texane/stlink/issues/651. In short, my USB 3.0 controller driver was buggy and I needed to update it.