platformio/platform-ststm32 #76: Arduino framework for STM32

sstaub, 2017-12-07:
Can you add the Arduino framework port from STM?
Announcement: [https://community.st.com/community/stm32-community/blog/2017/07/13/stm32-cores-enabled-in-arduino-ide](url)
Github: [https://github.com/stm32duino/Arduino_Core_STM32](url)

ArnieX, 2018-01-19:
Is this coming any time soon?

me21, 2018-03-19:
Isn't it already done?

ArnieX, 2018-03-19:
@me21 well not really, if I for example want to develop for Nucleo, these is option only for mbed framework not Arduino.

ubis, 2018-03-19:
It's in our plan, so it'll come. However, I don't have enough free time to dig in this, right now. 
Also I want to finish F1/F3/F4 board support of Maple core first. Oh, and contributions are welcome 😉

sstaub, 2018-03-29:
I think it make no really sense to support the old st framework. The new STM32 Arduino Core will be more important because there is also support for the nucleo add-on boards, Ethernet and FreeRtos.

ghent360, 2018-04-01:
+1 for this feature
@ubis - If I want to help where do I start?

ubis, 2018-04-04:
@ghent360 mostly reading `boards.txt`, `platform.txt` and then writing remap changes like it's done in [stm32f1.py](https://github.com/platformio/platform-ststm32/blob/develop/builder/frameworks/arduino/maple/stm32f1.py) and making a PR 🙂

me21, 2018-05-22:
Now that STM32F3 support has been dropped from rogerclark's Arduino core, I think it would be nice to include various Arduino cores into Platformio, namely:

- [Steve's Libmaple cores](https://github.com/stevstrong/Arduino_STM32) - seems to have some additional work on STM32F4 in comparison to Roger's cores already included in Platformio
- [Danieleff generic](https://github.com/danieleff/STM32GENERIC) - supports several STM32F7 boards
- [STM's official core](https://github.com/stm32duino/Arduino_Core_STM32) - this is what this issue started with

ubis, 2018-06-16:
Successfully ported for the F103C8/BluePill board. From now on, development will be on my [forked repo](https://github.com/ubis/platform-ststm32/tree/feature/add_stm32_arduino_core) and merged into `platform-ststm32` when it's ready.

Right know there's some manual work to-do in order to enable for the F103C8 board and I need to simplify everything so the changes could be pushed and tested by anyone.

xC0000005, 2018-06-27:
@me21, I used someone else's builder + a few modifications to get STM32GENERIC running fine with PIO. The biggest problem is that that core doesn't seem to be actively maintained, and there's a few problems with it. Other than figuring out how to get it to download, it wasn't too hard. You can set your development platform to point to my fork of the repo and your framework to arduinostm32generic and it should pull down what's needed and build fine.

nightvisi0n, 2018-06-28:
@ubis Thank you for your port. I would like to test it, but wasn't successful.
I changed the following things in my `platformio.ini`:

```ini
platform = https://github.com/ubis/platform-ststm32.git#feature/add_stm32_arduino_core
board = genericSTM32F103RB
build_flags=-D STM32_OFFICIAL_CORE
```
Unfortunately I only got `AssertionError [...] assert isdir(FRAMEWORK_DIR)`, which is reasonable since `.pioenvs/` is missing the `framework-arduinoststm32`. Using the latest release of the ststm32 platform everything works fine, so this should be related to your changes IMHO.
Do you have any idea what could be missing in my test setup?

ubis, 2018-06-28:
@nightvisi0n  No, it's on my end, because I haven't written anywhere how to use it, since some manual work needs to be done, in order to set up everything.

Now, it doesn't support generic boards at the moment, and we will discuss with Ivan about this, but in the meantime please use BluePill F103C8. Then create a folder called `STM32` in the `framework-arduinoststm32` folder and download [this](https://github.com/stm32duino/Arduino_Core_STM32) there. 
Now do the same for CMSIS, create `CMSIS` and download [this](https://github.com/ARM-software/CMSIS). 
Everything should be ready now and you can try to build it.

tve, 2018-08-04:
@ubis, I followed your directions in your last comment but I get the following error:

/home/tve/.platformio/packages/framework-arduinoststm32/STM32/system/Drivers/CMSIS/Device/ST/STM32L0xx/Include/stm32l053xx.h:129:10: fatal error: core_cm0plus.h: No such file or directory

In the gcc command line I see: `-I/home/tve/.platformio/packages/framework-arduinoststm32/STM32/CMSIS/CMSIS/Core/Include` but the `Core` directory seems to have vanished. Instead:
```
# find . -name core_cm0plus.h
./packages/framework-arduinoststm32/STM32/CMSIS/CMSIS/Include/core_cm0plus.h
```
I can track the source of that -I down, but am I on the right track?

ubis, 2018-08-04:
@tve Sorry, i made a typo and posted a link of CMSIS4 instead of CMSIS5. Please use this: [https://github.com/ARM-software/CMSIS_5 ](https://github.com/ARM-software/CMSIS_5). I've edited my previous post already.

You shouldn't need to edit anything, as everything should work out-of-the-box, except preparing that package with core + CMSIS.

tve, 2018-08-04:
Well, it doesn't "just work". I still get:

/home/tve/.platformio/packages/framework-arduinoststm32/STM32/cores/arduino/stm32/clock.c:39:10: fatal error: stm32yyxx_ll_cortex.h: No such file or directory

due to a missing `join(FRAMEWORK_DIR, "cores", "arduino", "stm32", "LL"),` in the builder script.

Also, the stuff in `syscalls_stm32.c` still doesn't get linked in.

ubis, 2018-08-04:
It seems that core has been updated a few days ago and made some changes about LL wrapper. I will update builder script now, thanks.

tve, 2018-08-04:
Yup, that does it! Thanks for the super-quick turn-around!

boristomas, 2018-09-07:
@ubis i still have issues with STM32L432.
I have tried it on 3 PC with fresh install of VS code (one is portable).
and it doesn't work
platformio.ini:
`[env:nucleo_l432kc]
platform = ststm32
framework = arduino
board = nucleo_l432kc
`
error:

> Processing nucleo_l432kc (platform: ststm32; board: nucleo_l432kc; framework: arduino)
-------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: ST STM32 > ST Nucleo L432KC
SYSTEM: STM32L432KCU6 80MHz 64KB RAM (256KB Flash)
DEBUG: CURRENT(stlink) ON-BOARD(stlink) EXTERNAL(blackmagic, jlink)
AssertionError: :
File "C:\Users\boris\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 169:
env.SConscript("$BUILD_SCRIPT")
File "C:\Users\boris\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
return _SConscript(self.fs, *files, **subst_kw)
File "C:\Users\boris\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
exec _file_ in call_stack[-1].globals
File "C:\Users\boris\.platformio\platforms\ststm32\builder\main.py", line 88:
target_elf = env.BuildProgram()
File "C:\Users\boris\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\Users\boris\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 107:
env.BuildFrameworks(env.get("PIOFRAMEWORK"))
File "C:\Users\boris\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\Users\boris\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 314:
SConscript(env.GetFrameworkScript(f), exports="env")
File "C:\Users\boris\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 604:
return method(*args, **kw)
File "C:\Users\boris\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
return _SConscript(self.fs, *files, **subst_kw)
File "C:\Users\boris\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
exec _file_ in call_stack[-1].globals
File "C:\Users\boris\.platformio\platforms\ststm32\builder\frameworks\arduino\stm32duino.py", line 56:
env.SConscript("stm32core.py")
File "C:\Users\boris\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
return _SConscript(self.fs, *files, **subst_kw)
File "C:\Users\boris\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
exec _file_ in call_stack[-1].globals
File "C:\Users\boris\.platformio\platforms\ststm32\builder\frameworks\arduino\stm32core.py", line 38:
assert isdir(FRAMEWORK_DIR)
The terminal process terminated with exit code: 1

ubis, 2018-09-07:
@boristomas Have you created `STM32` folder in `.platformio/packages/framework-arduinoststm32` and placed [this](https://github.com/stm32duino/Arduino_Core_STM32) inside it?

boristomas, 2018-09-07:
i created STM32, STM32L4,... you name it. error is the same

ubis, 2018-09-09:
I've changed main branch in my repo, so everyone can directly use it, instead of replacing platform files. However, package still needs to have `STM32` and `CMSIS` contents.  `platformio.ini` file looks like this:
```
[env:nucleo_l432kc]
platform = https://github.com/ubis/platform-ststm32.git
board = nucleo_l432kc
framework = arduino
```

@boristomas why I asked about STM32 folder, because error happens at 38 line, which is `assert isdir(FRAMEWORK_DIR)` and it fails because there is no `STM32` folder inside `framework-arduinoststm32`.

boristomas, 2018-09-09:
@ubis thank you for your effort, 
i have the same error with instructions you provided.
[image]
this is my `framework-arduinoststm32` folder

boristomas, 2018-09-09:
Ok, i think i got it, in .py it looks for CMSIS inside of STM32 what i did, and now it works!
I assume this process (creating folders and copying sources) will be automatic one day.
Thanks again.

vivian-ng, 2018-10-27:
Hi, has the Arduino_Core_STM32 package been ported over to Platformio? When I use ststm32 to compile for 
board = genericSTM32F407VET6
I get a lot of timer and watchdog related errors, like:
'IWDG_HandleTypeDef' does not name a type
'TIM_HandleTypeDef' does not name a type
'stm32f4_timer_t {aka struct tTimerConfig}' has no member named 'handle'

sstaub, 2018-10-27:
There is some development on supporting Arduino STM32Core, but I don't know when there is a release.

ubis, 2018-10-29:
Yes, there's official STM32 core support in develop branch. However, currently we support the exact boards (so different boards with same mcu's won't work, at the moment). Take a look in [here](https://github.com/stm32duino/Arduino_Core_STM32) to get a full supported board list. So in @vivian-ng's case, none of generic boards are supported, but `STM32F407G-DISC1` is, so use that instead.

ivankravets, 2018-12-11:
Thanks, @ubis and @valeros for adding support for official STm32 Core for Arduino.

We are going to make a public release this week. Please re-test the latest upstream version https://docs.platformio.org/en/latest/platforms/ststm32.html#upstream

Does it work for you?

sstaub, 2018-12-12:
Tested with Nucleo 429ZI, yes it works. Thank you very much.  👍

qu1ck, 2018-12-23:
Is there a way to explicitly specify which core I want to use? I have a project built using legacy (rogerclark's) core that is based on maple and upgrading stm32 platform to 4.6.0 caused a switch to new stm32 core, which broke my build.
For now I found a workaround: add
`board_build.variant = custom`
in `platformio.ini` but that workaround is dependent on logic in this file:
`https://github.com/platformio/platform-ststm32/blob/master/builder/frameworks/arduino.py`
which there is no guarantee that it won't change. I would prefer if there was explicit option to specify a core.

ivankravets, 2018-12-23:
Yes, if you need old version of dev/platform, please use `platform = ststm32@~4.5.0`

me21, 2018-12-23:
@ivankravets I hope your solution is only temporary, because using older version of the platform cuts us off future updates of core and compiler.

qu1ck, 2018-12-24:
@Alteregoxxx see this commit:
https://github.com/openscopeproject/HP34401a-OLED-FW/commit/d7cab8e4ad88a5425c79168f777df9af244f10e9

It works right now because ststm32 platform build file (linked it above) checks if the board variant is one of the explicit variants that are configured in the framework-arduinoststm32 package. Since "custom" isn't one of them, it reverts to legacy (unless board mcu is one of stm32f1 or stm32f4 in which case there is a hail mary fallback to new stm32 core).

I am aware that you can "pin" the version of the platform for the project in ini file but that is unworkable solution. If I have more than 1 projects using different cores I can't keep downloading different versions all the time when I switch between them.

sstaub, 2018-12-24:
Maybe the Maple boards and clones should have a seperate platform.

me21, 2018-12-24:
Can it be made possible to choose core (official or rogerclarkmelbourne) in the platform?

ivankravets, 2018-12-24:
@valeros is it possible to add support for `-D PIO_FRAMEWORK_ARDUINO_LEGACY` (forcing build process to `rogerclarkmelbourne` core )?

aster94, 2019-01-03:
> Now that STM32F3 support has been dropped from rogerclark's Arduino core, I think it would be nice to include various Arduino cores into Platformio, namely:
>
> * [Steve's Libmaple [...]

agree with this
any of them have pro and cons. It isn't a good idea to choose just one, for example:
* [Steve's/Roger Libmaple cores] for now this is the best option but not always perfectly compliant with the arduino API
 * [Danieleff generic] not very active
 * [STM's official core] maybe it would be the best in the next couple of years but for now it lack of various features

so the possibility to choose which one to use would be very useful :)

rzr, 2019-01-05:
@fpistm, I suppose that f767zi will get arduino support in next stm32diuno release ? or should I patch anything ?

hasenbanck, 2019-01-05:
@rzr The ["official" core](https://github.com/stm32duino/Arduino_Core_STM32) already supports the Nucleo F767ZI board in the current version 1.4

rzr, 2019-01-05:
So I should add arduino framework to platformio board defs, until 1.5 is released (with a couple of patches from me)

sstaub, 2019-01-05:
Yes, please add the Nucleo-F767ZI to the board defs. I got a board last week and so I'm missing the support. :)

ivankravets, 2019-02-21:
> Yes, please add the Nucleo-F767ZI to the board defs

Added, see https://github.com/platformio/platform-ststm32/blob/develop/boards/nucleo_f767zi.json

> until 1.5 is released

Added support for 1.5 in the latest upstream version of dev/platform. See docs https://docs.platformio.org/en/latest/platforms/ststm32.html#stable-and-upstream-versions

Please re-test.

P.S: Switching between Arduino cores: https://docs.platformio.org/en/latest/platforms/ststm32.html#switching-between-arduino-cores

closed 2019-02-21 by ivankravets

andhieSetyabudi, 2019-09-11:
has anybody try to use ladybug-L432KC in platformio??

even the mcu is same with nucleo-L432kc but i can't upload my .hex to board..

thanks