platformio/platform-ststm32 #78: Nucleo F103RB issue with default linker file

ELuigi, 2017-12-10:
Hello,
I have an issue with the default linker file of my nucleo STM32F103RB. When I create a simple blink program for this board with following platform.ini:
[env:nucleo_f103rb]
platform = ststm32
board = nucleo_f103rb
framework = stm32cube
I have no issue during build or upload but the led doesn't blink and impossible to debug the program on the board. To fix it I have modify the file STM32F103RB_DEFAULT.ld by changing below bolds values (seen in another F103RB's ld file of another IDE)
/* Highest address of the user mode stack */ 
_estack = **0x20005000**;    /* end of RAM */ previous was 0x2005000
/* Generate a link error if heap and stack don't fit into RAM */
_Min_Heap_Size = **0x0**;      /* required amount of heap  */ previous was 0x200

now I am able to see the led blinked and debug the program.

Could you please upload the framework to take this modification into account ? or tell me where modify it (and I will do a pull request) because I have just seen a python script which generate default file?

And thank you for your very very great job with platformio !!!!

Etienne

ivankravets, 2017-12-10:
@valeros could you help here?

ELuigi, 2017-12-14:
Hi @valeros! I have followed your instruction and that works !
Thank you very much for your help

closed 2017-12-14 by ELuigi