platformio/platform-ststm32 #82: bluepill and stm32cube doesn't work.

ikeji, 2018-01-04:
I did follow step:
1. download and extract https://github.com/platformio/platform-ststm32/archive/develop.zip
2. cd platform-ststm32-develop/examples/stm32cube-hal-blink
3. add follow code to platformio.ini
----
[env:bluepill_f103c8]
platform = ststm32
framework = stm32cube
board = bluepill_f103c8
build_flags = -DF1
----
4. pio run -e bluepill_f103c8 --target upload

But the LED doesn't flash.

I tried to debug by gdb.
It moved to HardFault_Handler when __libc_init_array is invoked.
----
(gdb) disassemble 
Dump of assembler code for function LoopFillZerobss:
   0x080005fe <+0>:     ldr     r3, [pc, #36]   ; (0x8000624 <LoopFillZerobss+38>)
   0x08000600 <+2>:     cmp     r2, r3
   0x08000602 <+4>:     bcc.w   0x80005f8 <FillZerobss>
   0x08000606 <+8>:     bl      0x800062c <SystemInit>
=> 0x0800060a <+12>:    bl      0x8000160 <__libc_init_array>
   0x0800060e <+16>:    bl      0x80002a8 <main>
(snip)
End of assembler dump.
(gdb) stepi
0x08000160 in __libc_init_array ()
(gdb) disassemble 
Dump of assembler code for function __libc_init_array:
=> 0x08000160 <+0>:     push    {r4, r5, r6, lr}
   0x08000162 <+2>:     ldr     r6, [pc, #60]   ; (0x80001a0 <__libc_init_array+64>)
   0x08000164 <+4>:     ldr     r5, [pc, #60]   ; (0x80001a4 <__libc_init_array+68>)
(snip)
End of assembler dump.
(gdb) si
HardFault_Handler () at src/main.c:59
59      {
(gdb) 
----
I guess for some reason it failed to setup stack.

ubis, 2018-02-05:
There were some minor changes regarding BluePill board and STM32Cube framework, although it was only the use of correct ldscript for F103C8 mcu.
Could you re-check with latest develop build?

Kumi27, 2018-02-15:
There is an error in default linker script for the STM32F103C8 chip. 
In the STM32F103C8_DEFAULT.ld file the line for stack has a missing “0”
`_estack = 0x2005000; /* end of RAM */`
and the correct value is
`_estack = 0x20005000; /* end of RAM */`
After change and recompilation, the BluePill started to blink.
[](https://community.platformio.org/t/stm32cube-and-bluepill-f103c8-no-linker-script-no-boot/2929/2)

ivankravets, 2018-02-21:
Fixed recently

closed 2018-02-21 by ivankravets