platformio/platform-ststm32 #83: Add support for STM32CubeProgrammer

ivankravets, 2018-01-13:
https://my.st.com/content/my_st_com/en/products/development-tools/software-development-tools/stm32-software-development-tools/stm32-programmers/stm32cubeprog.license=1515792898204.html

Phando, 2020-04-20:
Is it possible to use the platformio.ini to point the builder at the STM32 Cube Programmer?

angelnu, 2020-06-02:
it worked for me adding the following lines to my platformio.ini
```
[env:nucleo_test]
  platform = ststm32
  board = nucleo_l412kb
  framework = arduino
  ;upload_protocol = serial ;Use this for serial
  ;UPLOAD_PORT=SWD ;Use this for ST-LINK
  upload_command =
   /Applications/STMicroelectronics/STM32Cube/STM32CubeProgrammer/STM32CubeProgrammer.app/Contents/MacOs/bin/STM32_Programmer_CLI -c port=$UPLOAD_PORT -w $SOURCE 0x8000000

```

I did this after the nucleo_l412kb was failing for serial and I manually had to update openocd to get support for the l412kb. It is working well in my Mac.