platformio/platform-ststm32 #86: Add support for RAK811 board

Ebiroll, 2018-02-10:
Please add something like this, rak811.json
STM32L161 mcu.
https://www.aliexpress.com/store/product/RAK811-LoRa-Tracker-Board-MAX-7Q-GPS-Module-and-MEMS-Sensor-Wireless-Remote-Positioning-Solution-LoRaWAN/2805180_32844470946.html?spm=2114.12010608.0.0.219c103eDASa9w

I think earlier versions of the board only had 16K ram so stm32l15xx6.ld is maybe better as ldscript.
{
    "build": {
        "core": "stm32",
        "cpu": "cortex-m3",
        "extra_flags": "-DSTM32L151xB -DSTM32L151CBU6",
        "hwids": [
            [
                "0x0429",
                "0x0003"
            ],
            [
                "0x0429",
                "0x0004"
            ]
        ],
        "f_cpu": "32768000L",
        "mcu": "stm32l151cbu6",
        "variant": "stm32l151xba",
        "ldscript": "/home/olof/rak/STM32L151XBA_FLASH.ld"
    },
    "frameworks": [
        "mbed",
        "cmsis"
    ],
    "name": "Rak811",
    "upload": {
        "maximum_ram_size": 32768,
        "maximum_size": 128000
    },
    "url": "http://www.rakwireless.com/en/download/RAK811/Hardware%20Design",
    "vendor": "RAK"
}

Linker script, https://github.com/RAKWireless/RAK811_BreakBoard/blob/master/src/boards/RAK811BreakBoard/cmsis/arm-gcc/STM32L151XBA_FLASH.ld

stm32flash /dev/ttyUSB0
stm32flash 0.5
http://stm32flash.sourceforge.net/

Interface serial_posix: 57600 8E1
Version      : 0x31
Option 1     : 0x00
Option 2     : 0x00
Device ID    : 0x0429 (STM32L1xxx6(8/B)A)
- RAM        : 32KiB  (4096b reserved by bootloader)
- Flash      : 128KiB (size first sector: 16x256)
- Option RAM : 32b
- System RAM : 4KiB

Here is source,
https://github.com/RAKWireless/RAK811_BreakBoard

sabas1080, 2018-03-26:
I am trying to port the RAK811 to the Arduino Core, but I am having some problems wanting to use the module Lora with the Lmic (https://github.com/mcci-catena/arduino-lmic) or Lora (https://github.com/sandeepmistry/arduino-LoRa) libraries with pins RADIO_RF_CRX_RX  PB_6  //CRF3 (ANT_SWITCH_TX_BOOST),  RADIO_RF_CBT_HF                             PB_7  //CRF2 HF (ANT_SWITCH_TX_RFO,  RADIO_RF_CTX_PA , any help is welcome

https://github.com/stm32duino/Arduino_Core_STM32/pull/229

ivankravets, 2019-03-05:
Resolved in https://github.com/platformio/platform-ststm32/commit/f37b7c13b4520cc2c90392687eece20676ed43eb

closed 2019-03-05 by ivankravets