platformio/platform-ststm32 #87: bluepill_f103c8 wont work with dfu upload and STM32duino-bootloader

zerog2k, 2018-02-24:
looks possibly similar to #53 
using a generic bluepill board, with this bootloader 
https://github.com/rogerclarkmelbourne/STM32duino-bootloader/blob/master/binaries/generic_boot20_pc13.bin

not able to get functional code with this:

```
[env:bluepill_f103c8]
platform = ststm32
framework = arduino
board = bluepill_f103c8
upload_port = /dev/ttyACM2
upload_protocol = dfu
```
but i do get functional code from this:
```
[env:genericSTM32F103C8]
platform = ststm32
framework = arduino
board = genericSTM32F103C8
upload_protocol = dfu
upload_port = /dev/ttyACM2
```

zerog2k, 2018-02-24:
here is the pio output from bluepill_f103c8:
```
[Sat Feb 24 17:37:50 2018] Processing bluepill_f103c8 (platform: ststm32; board: bluepill_f103c8; framework: arduino)
-------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: ST STM32 > BluePill F103C8
SYSTEM: STM32F103C8T6 72MHz 20KB RAM (64KB Flash)
DEBUG: CURRENT(blackmagic) EXTERNAL(blackmagic, jlink, stlink)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(light)
Collected 28 compatible libraries
Scanning dependencies...
No dependencies
Linking .pioenvs/bluepill_f103c8/firmware.elf
Checking program size
text       data     bss     dec     hex filename
6984       1944     344    9272    2438 .pioenvs/bluepill_f103c8/firmware.elf
Configuring upload protocol...
AVAILABLE: blackmagic, dfu, jlink, mbed, stlink
CURRENT: upload_protocol = dfu
Looking for upload port...
Use manually specified: /dev/ttyACM2
Uploading .pioenvs/bluepill_f103c8/firmware.bin
Failed to open serial device.
dfu-util 0.7

Copyright 2005-2008 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2012 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

Filter on vendor = 0x1eaf product = 0x0003
Opening DFU capable USB device... ID 1eaf:0003
Run-time device DFU version 0110
No valid DFU suffix signature
Found DFU: [1eaf:0003] devnum=0, cfg=1, intf=0, alt=2, name="STM32duino bootloader v1.0  Upload to Flash 0x8002000"
Claiming USB DFU Interface...
Warning: File has no DFU suffix
Setting Alternate Setting #2 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0110
Device returned transfer size 1024
bytes_per_hash=178
Copying data from PC to DFU device
Starting download: [##################################################] finished!
state(8) = dfuMANIFEST-WAIT-RESET, status(0) = No error condition is present
Done!
Resetting USB to switch back to runtime mode
=========================================== [SUCCESS] Took 4.52 seconds ===========================================

==================================================== [SUMMARY] ====================================================
Environment bluepill_f103c8     [SUCCESS]
Environment maple               [SKIP]
Environment genericSTM32F103C8  [SKIP]
=========================================== [SUCCESS] Took 4.52 seconds ===========================================

Terminal will be reused by tasks, press any key to close it.
```

here is the output from genericSTM32F103C8:
```
[Sat Feb 24 17:39:04 2018] Processing genericSTM32F103C8 (platform: ststm32; board: genericSTM32F103C8; framework:arduino)
-------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: ST STM32 > STM32F103C8 (20k RAM. 64k Flash)
SYSTEM: STM32F103C8T6 72MHz 20KB RAM (64KB Flash)
DEBUG: CURRENT(blackmagic) EXTERNAL(blackmagic, jlink, stlink)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(light)
Collected 28 compatible libraries
Scanning dependencies...
No dependencies
Configuring upload protocol...
AVAILABLE: blackmagic, dfu, jlink, serial, stlink
CURRENT: upload_protocol = dfu
Looking for upload port...
Use manually specified: /dev/ttyACM2
Uploading .pioenvs/genericSTM32F103C8/firmware.bin
Failed to open serial device.
No valid DFU suffix signature
dfu-util 0.7
Warning: File has no DFU suffix

Copyright 2005-2008 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2012 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

Filter on vendor = 0x1eaf product = 0x0003
Opening DFU capable USB device... ID 1eaf:0003
Run-time device DFU version 0110
Found DFU: [1eaf:0003] devnum=0, cfg=1, intf=0, alt=2, name="STM32duino bootloader v1.0  Upload to Flash 0x8002000"
Claiming USB DFU Interface...
Setting Alternate Setting #2 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0110
Device returned transfer size 1024
bytes_per_hash=285
Copying data from PC to DFU device
Starting download: [##################################################] finished!
state(8) = dfuMANIFEST-WAIT-RESET, status(0) = No error condition is present
Done!
Resetting USB to switch back to runtime mode
=========================================== [SUCCESS] Took 2.96 seconds ===========================================

==================================================== [SUMMARY] ====================================================
Environment bluepill_f103c8     [SKIP]
Environment maple               [SKIP]
Environment genericSTM32F103C8  [SUCCESS]
=========================================== [SUCCESS] Took 2.97 seconds ===========================================

Terminal will be reused by tasks, press any key to close it.
```

noticed that bluepill seems to have an extra linking step during upload?
```
Linking .pioenvs/bluepill_f103c8/firmware.elf
Checking program size
text       data     bss     dec     hex filename
6984       1944     344    9272    2438 .pioenvs/bluepill_f103c8/firmware.elf
```

zerog2k, 2018-02-24:
ok the "extra" linking step seems to happen again during upload step whenever i change environments (boards), even after I did a compile step prior to upload step. Not sure why.

zerog2k, 2018-02-24:
the prog size does look significantly larger with generic than bluepill:

text       data     bss     dec     hex filename
12100      2152     976   15228    3b7c .pioenvs/genericSTM32F103C8/firmware.elf

vs

Linking .pioenvs/bluepill_f103c8/firmware.elf
Checking program size
text       data     bss     dec     hex filename
6984       1944     344    9272    2438 .pioenvs/bluepill_f103c8/firmware.elf

I wonder why this would be. The code did not change.. only change was environment targeted (i.e. the board type)

zerog2k, 2018-02-25:
hmm looks like if board name contains "generic" and upload type is "dfu" then there is some different linking happening...
https://github.com/platformio/platform-ststm32/blob/develop/builder/frameworks/arduino/maple/stm32f1.py#L69-L79

I wonder if "bluepill_f103c8" is not what I'm thinking it is?

ivankravets, 2018-03-02:
Fixed in https://github.com/platformio/platform-ststm32/commit/5da2246b00c5a60fd7c3469af8d74436f66a2c24

Please re-test

closed 2018-03-02 by ivankravets