platformio/platform-ststm32 #89: STM32CubeMX/HAL RNG/RCC compile errors

cchamilt, 2018-02-25:
Hi, I have platformio building a CubeMX simple version project.  However, certain components seem to be unsupported somehow.

Specifically, if I enable the RNG, I then have to manually add
```
#include "stm32f4xx_hal_rng.h"
```
in main.c to compile.

RCC however fails even when I add
```
#include "stm32f4xx_hal_rcc_ex.h"
```
to main.c.

Is there a good way I can help the 'magic' to work?

```
Compiling .pioenvs\fieroMBrevA\src\gpio.c.o
Compiling .pioenvs\fieroMBrevA\src\main.c.o
Compiling .pioenvs\fieroMBrevA\src\rtc.c.o
CCompiling .pioenvs\fieroMBrevA\src\stm32f4xx_hal_msp.c.o
Src\main.c: In function 'SystemClock_Config':
Src\main.c:167:64: error: 'RCC_PERIPHCLK_CLK48' undeclared (first use in this function); did you mean 'RCC_PERIPHCLK_CK48'?

```
configuration:
```
[platformio]
env_default = fieroMBrevA
include_dir = Inc
src_dir = Src

[env:fieroMBrevA]
platform = ststm32
lib_ldf_mode = deep+
framework = stm32cube
board = stm32fiero
upload_protocol = gdb
build_flags = -DF4 -DUSE_HAL_DRIVER -DSTM32F413x -IInc 
lib_extra_dirs=Inc/
```

valeros, 2018-12-11:
Probably you shouldn't include `stm32f4xx_hal_rcc_ex.h` explicitly. You can try to install ststm32 platform from dev branch and include only `stm32f4xx_hal.h`,

valeros, 2020-11-26:
Please reopen if you still need help.

closed 2020-11-26 by valeros