platformio/platform-ststm32 #90: again: undefined reference to `_sbrk'

LexNetAT, 2018-03-01:
I have given it up after many hours, because I always get this error message:
> ~/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/lib/thumb/v7-m\libc.a(lib_a-sbrkr.o):
>  In function `_sbrk_r`:
> sbrkr.c:(.text._sbrk_r+0xc): undefined reference to `_sbrk`
> collect2.exe: error: ld returned 1 exit status
> *** [.pioenvs\bluepill_f103c8\firmware.elf] Error 1

My `platform.ini` looks like this:
> [env:bluepill_f103c8]
> platform = ststm32
> board = bluepill_f103c8
> framework = arduino
> build_flags =
>    -std=gnu++11
>    -DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG=1
>    -DSERIAL_USB
>    -DGENERIC_BOOTLOADER

Apparently this mistake was already there and now it seems to be back.
The already solved issues didn't help me.
I hope this problem does not just come back to me.

LexNetAT, 2018-03-01:
I have now found a preliminary solution -> take older version.
Result of my experiments:
- current release 4.0.1 wont work
- version before 4.0.0 also wont work
- version 3.8.0 still work (`platform.ini` has now entry `platform = ststm32@3.8.0`)

ivankravets, 2018-03-02:
@ubis do you have any ideas?

me21, 2018-03-02:
Temporary workaround: add the following snippet to one of your source files and rebuild:

```
extern "C" void *_sbrk(int incr);
void dummy_sbrk_caller() __attribute__((__used__));
void dummy_sbrk_caller()
{
  _sbrk(0);
}
```

Explanation is [here](https://stackoverflow.com/a/28895983/2404492).

ubis, 2018-03-02:
Exactly what @me21 said about `libc`. 
Before, we had double `c` definitions. When I did refactoring, I couldn't reproduce this problem and I was hoping that it was fixed. So I've removed one of `c` definitions in linker flags. Yeah, my bad..

commit 7977e787 (2018-03-02): Fix "undefined reference to `_sbrk'" // Resolve #90
closed 2018-03-02 by ivankravets