platformio/platform-ststm32 #92: Setting own upload command

danielkucera, 2018-03-08:
https://github.com/platformio/platform-ststm32/blob/a0fa57381ef97022474f3e0320ebf194cba9784b/builder/main.py#L214

How does this section works? How can I set UPLOADCMD? I tried:
```
export UPLOADCMD="echo BOOOOOM!"
```
adding into platformio.ini:
```
UPLOADCMD = "echo AHOJ"
uploadcmd = "echo AHOJ"
```

but still nothing.

ivankravets, 2018-03-08:
http://docs.platformio.org/en/latest/projectconf/advanced_scripting.html

closed 2018-03-08 by ivankravets

danielkucera, 2018-03-08:
Ok, thank you, I've figured it out, but it's not exactly elegant solution:
https://github.com/danielkucera/stm32-ir-keyboard/commit/7a501d5e50e8a7735c3b7a9732bc6f87915bed41
It would be nice if the  UPLOADCMD could be set directly in platformio.ini