platformio/platform-ststm32 #93: Binaries created for Nucleo F401re with mbed do not flash succesfully

cocoaway, 2018-03-14:
On macOS,  stm32 Nucleo F401re blink led compiled, flashed on mbed disk but non executed.

Copying the firmware manually produces the same result.
The file is transferred to the STLink, but the STLink does not flash the firmware to the MCU or apparently respond in any way.

Same example compiled from mbed online successfully runs.

andrewadare, 2018-04-05:
I encountered exactly the problem described in the link above on macOS with the Nucleo L432KC. I am using version PlatformIO version 3.5.3a8.

andrewadare, 2018-04-14:
After reading [this post](https://os.mbed.com/forum/bugs-suggestions/topic/28258/) on the MBED forum, I tried updating the STLink firmware on my Nucleo board. After that, flashing works. This may be worth a shot on the F401re.

ivankravets, 2018-06-04:
https://github.com/platformio/platform-ststm32/issues/93#issuecomment-381344545

closed 2018-06-04 by ivankravets